markusressel/fan2go #262: Provide a Flatpak package.

RokeJulianLockhart, 2023-10-01:
**Is your feature request related to a problem? Please describe.**
I don't want to have to compile this.

**Describe the solution you'd like**
A package which works anywhere, since that's less work on your part.

**Describe alternatives you've considered**
Native packages, like `.rpm` and `.deb`, or perhaps a Snap package. The former are ideal.

**Additional context**
Snap would also provide this, but it has issues. Native packages would be best, although that's slightly more work (perhaps unless you use SUSE's Open Build Service?)

markusressel, 2023-10-01:
Hey @RokeJulianLockhart again,

As I cannot provide support for systems I don't use, I only added a package for the AUR and do not intend to add any other packages to the list. If you want to create a package for fan2go on a different system feel free to do so, I will include (simple) installation instructions in the README. This has already happened for Nix OS, see: https://github.com/markusressel/fan2go/issues/261#issuecomment-1742231777

RokeJulianLockhart, 2023-10-02:
The moment I'm able to, I'll do so.

markusressel, 2024-02-27:
Closing this for now due to inactivity.
Feel free to reopen or create a PR if you want to add something to the README.
Thx.

closed 2024-02-27 by markusressel

RokeJulianLockhart, 2024-05-24:
> https://github.com/markusressel/fan2go/issues/262#issuecomment-1965627987

@markusressel, https://github.com/markusressel/fan2go/releases/download/0.8.1/fan2go-linux-amd64 actually appears to provide what I want (albeit not exactly what this issue requests) - a compiled binary, so that I need not compile it myself.

markusressel, 2024-05-24:
See #269