markusressel/fan2go #264: Fails to set PWM value automatically, even though it does this manually

mks-h, 2023-10-22:
**Describe the bug**
When running fan2go as a daemon, it fails to set PWM value, reporting `WARNING  PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128`, and the speed doesn't change. Yet, it doesn't have issues changing it with `sudo /usr/local/bin/fan2go fan -i cpu_fan speed 255`.

Just to be clear, there is no third party controlling the fan, and I can control it manually just fine:
```
# cat /sys/class/hwmon/hwmon3/pwm1 
255
# echo 128 > /sys/class/hwmon/hwmon3/pwm1 
# cat /sys/class/hwmon/hwmon3/pwm1 
128
```

**Desktop (please complete the following information):**

- Distro: OpenSUSE MicroOS (Tumbleweed base)
- `uname -a`: Linux Vostro 6.5.6-1-default #1 SMP PREEMPT_DYNAMIC Fri Oct  6 11:20:48 UTC 2023 (c97c2df) x86_64 x86_64 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: 0.8.1 (compiled from 17451b6afa59752a1519931b8cc54de8d28a7e00)

**Additional context**
<details>
<summary>Config</summary>

```
dbPath: "/etc/fan2go/fan2go.db"

fans:
  - id: cpu_fan
    hwmon:
      platform: dell_smm-isa-0000
      rpmChannel: 1
      pwmChannel: 1
    neverStop: true
    curve: cpu_curve
    minPwm: 128
    startPwm: 128
    maxPwm: 255
    pwmMap:
      0: 0
      64: 128
      128: 255

sensors:
  - id: sensor1
    hwmon:
      platform: dell_smm-isa-0000
      index: 1
  - id: sensor2
    hwmon:
      platform: dell_smm-isa-0000
      index: 2
  - id: sensor3
    hwmon:
      platform: dell_smm-isa-0000
      index: 3

curves:
  - id: cpu_curve
    function:
      type: maximum
      curves:
        - curve1
        - curve2
        - curve3
  - id: curve1
    linear:
      sensor: sensor1
  - id: curve2
    linear:
      sensor: sensor2
  - id: curve3
    linear:
      sensor: sensor3

statistics:
  enabled: false

api:
  enabled: false

profiling:
  enabled: false
```

</details>

markusressel, 2023-10-22:
This is expected, since you specified:

```
pwmMap:
      0: 0
      64: 128
      128: 255
```

This tells fan2go that **setting** a PWM value of `128` will result in a PWM **readout** of 255. Therefore fan2go expects to read `255` after setting a value of `128`, but since it reads something different (128 in this case) a warning is printed.

What was your intention with specifying this `pwmMap` configuration?

mks-h, 2023-10-22:
This happens without the `pwmMap`, too. Here:
```
INFO  PWM settings of fan 'cpu_fan': Min 128, Start 128, Max 255
 INFO  Starting controller loop for fan 'cpu_fan'
 WARNING  PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
```

My fan only actually supports `0`, `128`, and `255` values, so I was messing around trying to specify this somehow to possibly fix the warning.

mks-h, 2023-10-22:
P.S. The warning constantly repeats

markusressel, 2023-10-22:
fan2go will always try to detect a valid `pwmMap` configuration automatically. When specifying a custom one in the configuration, the manual definition will override the automatically detected one.

Also, the automatically detected `pwmMap` is only computed once and then saved to the db file for later use. Removing the manual definition will cause the saved map to be used again. You can turn on verbose logging to see the contents of the `pwmMap` when the daemon starts up and see if it contains a valid configuration.

If not, overriding it in the config is the correct solution. I would guess something like this may work:

```
  pwmMap:
      0: 0
      128: 128
      255: 255
```

Note: This PWM map is currently **not** taken into account when using the `fan speed` command.

mks-h, 2023-10-22:
OK, so I've just deleted the db file, and ran command `sudo /usr/local/bin/fan2go -c /etc/fan2go/fan2go.yaml --no-style` (the one from the service file, but directly with sudo):
```
fan2go
INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu_fan...
INFO: Loading fan curve data for fan 'cpu_fan'...
WARNING: Fan 'cpu_fan' has not yet been analyzed, starting initialization sequence...
INFO: Computing pwm map...
INFO: Measuring RPM curve...
INFO: FanController: Using saved value for pwm map of Fan 'cpu_fan'
INFO: PWM settings of fan 'cpu_fan': Min 128, Start 128, Max 255
INFO: Starting controller loop for fan 'cpu_fan'
```

And it seem to work... For this run. When I close it (Ctrl+C) and run again — it makes the warning.
Closing the previous one:
```
^CINFO: Received SIGTERM signal, exiting...
INFO: Stopping RPM monitor of fan controller for fan cpu_fan...
INFO: Stopping fan controller for fan cpu_fan...
INFO: Trying to restore fan settings for cpu_fan...
INFO: Stopping sensor monitor for sensor sensor3...
INFO: Sensor Monitor for sensor sensor3 stopped.
INFO: Stopping sensor monitor for sensor sensor2...
INFO: Sensor Monitor for sensor sensor2 stopped.
INFO: Stopping sensor monitor for sensor sensor1...
INFO: Sensor Monitor for sensor sensor1 stopped.
INFO: Fan controller for fan cpu_fan stopped.
INFO: Done.
```
Starting again:
```
fan2go
INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu_fan...
INFO: Loading fan curve data for fan 'cpu_fan'...
INFO: FanController: Using saved value for pwm map of Fan 'cpu_fan'
INFO: PWM settings of fan 'cpu_fan': Min 128, Start 128, Max 255
INFO: Starting controller loop for fan 'cpu_fan'
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
WARNING: PWM of cpu_fan was changed by third party! Last set PWM value was: 255 but is now: 128
^CINFO: Received SIGTERM signal, exiting...
INFO: Stopping sensor monitor for sensor sensor3...
INFO: Sensor Monitor for sensor sensor3 stopped.
INFO: Stopping sensor monitor for sensor sensor1...
INFO: Sensor Monitor for sensor sensor1 stopped.
INFO: Stopping RPM monitor of fan controller for fan cpu_fan...
INFO: Stopping sensor monitor for sensor sensor2...
INFO: Sensor Monitor for sensor sensor2 stopped.
INFO: Stopping fan controller for fan cpu_fan...
INFO: Trying to restore fan settings for cpu_fan...
INFO: Fan controller for fan cpu_fan stopped.
INFO: Done.
```

When trying to run as a systemd service (the very same command), with or without a db, it alsways just crashes:
```
× fan2go.service - Advanced Fan Control program
     Loaded: loaded (/etc/systemd/system/fan2go.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sun 2023-10-22 22:09:32 EEST; 10s ago
   Duration: 31ms
    Process: 3992 ExecStart=/usr/local/bin/fan2go -c /etc/fan2go/fan2go.yaml --no-style (code=exited, >
   Main PID: 3992 (code=exited, status=1/FAILURE)
        CPU: 19ms

Oct 22 22:09:32 Vostro systemd[1]: fan2go.service: Scheduled restart job, restart counter is at 5.
Oct 22 22:09:32 Vostro systemd[1]: fan2go.service: Start request repeated too quickly.
Oct 22 22:09:32 Vostro systemd[1]: fan2go.service: Failed with result 'exit-code'.
Oct 22 22:09:32 Vostro systemd[1]: Failed to start Advanced Fan Control program.
```

mks-h, 2023-10-22:
>  I would guess something like this may work:

But using your `pwmMap` config makes it work when running manually, thanks. I guess everything else are different problems.

mks-h, 2023-10-22:
OK, yeah, the systemd issue is just selinux blocking it. Thanks for the help!

closed 2023-10-22 by mks-h