markusressel/fan2go #265: Inappropriately using the bug tracker to figure out a configuration issue that I am having and not understanding how to fix it

thyTwilightGoth, 2023-11-25:
Hi here is my fan2go.yaml config:

```
fans:
    - id: CPU
      cmd:
        setPwm:
          exec: /usr/bin/nbfc
          args: ["set", "-f", "0", "-s", "%pwm%"]

        getRpm:
          exec: /usr/bin/cat
          args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan1_input"]

        minPwm: 12.5
        startPwm: 12.5
        maxPwm: 100

    - id: GPU
      cmd:
        setPwm:
          exec: /usr/bin/nbfc
          args: ["set", "-f", "1", "-s", "%pwm%"]

        getRpm:
          exec: /usr/bin/cat
          args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan2_input"]

        minPwm: 12.5
        startPwm: 12.5
        maxPwm: 100

sensors:
    - id: CPU_Temp
      hwmon:
        platform: k10temp-pci-00c3
        index: 1

    - id: GPU_Temp
      hwmon:
        platform: amdgpu-pci-0c00
        index: 1

curves:
    - id: CPU_Curve
      linear:
        sensor: CPU_Temp
        min: 40
        max: 80

    - id: GPU_Curve
      linear:
        sensor: GPU_Temp
        min: 40
        max: 80
```

And here is the output of ```doas fan2go config validate```

```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 WARNING  Unused curve configuration: cpu
 WARNING  Unused curve configuration: GPU_Curve
  ERROR   Validation failed: fan CPU: missing curve definition in configuration entry
```

I am on arch linux

jhillest, 2023-11-25:
Each sequence (fans, sensors, curves) are indented 4 spaces, but the key and value pairs are indented by 2 spaces.
The package `yamllint`, available in the extra repository, will complain of inconsistent use of indentation on your config.

thyTwilightGoth, 2023-11-25:
Well, I got a different error when I the indentation in my config to this:

```
fans:
    - id: CPU
        cmd:
        setPwm:
            exec: /usr/bin/nbfc
            args: ["set", "-f", "0", "-s", "%pwm%"]

        getRpm:
            exec: /usr/bin/cat
            args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan1_input"]

        minPwm: 12.5
        startPwm: 12.5
        maxPwm: 100

    - id: GPU
        cmd:
        setPwm:
            exec: /usr/bin/nbfc
            args: ["set", "-f", "1", "-s", "%pwm%"]

        getRpm:
            exec: /usr/bin/cat
            args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan2_input"]

        minPwm: 12.5
        startPwm: 12.5
        maxPwm: 100

sensors:
    - id: CPU_Temp
        hwmon:
            platform: k10temp-pci-00c3
            index: 1

    - id: GPU_Temp
        hwmon:
            platform: amdgpu-pci-0c00
            index: 1

curves:
    - id: CPU_Curve
        linear:
            sensor: CPU_Temp
            min: 40
            max: 80

    - id: GPU_Curve
        linear:
            sensor: GPU_Temp
            min: 40
            max: 80
```

The error now is:

```
  ERROR   Error sending notification: exit status 1
  FATAL   Error reading config file, While parsing config: yaml: line 3: mapping values are not allowed in this context
```

I don't know what I did wrong, but at least the error is different now T_T

jhillest, 2023-11-25:
Did you install `yamllint` and ran that on your config?

thyTwilightGoth, 2023-11-25:
No I didn't, I erroneously thought that was part of the validate command. I have know and here is the output:

```
/etc/fan2go/fan2go.yaml
  1:1       warning  missing document start "---"  (document-start)
  3:12      error    syntax error: mapping values are not allowed here (syntax)

```

jhillest, 2023-11-25:
The following YAML-file gives no errors.
I also changed the indentation from 4 spaces to 2 spaces to cut down the lenght of the lines.

`---
fans:
  - id: CPU
    cmd:
      setPwm:
        exec: /usr/bin/nbfc
        args: ["set", "-f", "0", "-s", "%pwm%"]

      getRpm:
        exec: /usr/bin/cat
        args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan1_input"]

      minPwm: 12.5
      startPwm: 12.5
      maxPwm: 100

  - id: GPU
    cmd:
      setPwm:
        exec: /usr/bin/nbfc
        args: ["set", "-f", "1", "-s", "%pwm%"]

      getRpm:
        exec: /usr/bin/cat
        args: ["/sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/fan2_input"]

      minPwm: 12.5
      startPwm: 12.5
      maxPwm: 100

sensors:
  - id: CPU_Temp
    hwmon:
      platform: k10temp-pci-00c3
      index: 1

  - id: GPU_Temp
    hwmon:
      platform: amdgpu-pci-0c00
      index: 1

curves:
  - id: CPU_Curve
    linear:
      sensor: CPU_Temp
      min: 40
      max: 80

  - id: GPU_Curve
    linear:
      sensor: GPU_Temp
      min: 40
      max: 80
`
There was incorrect indententation of `setPwm` and `getPwm` under the `cmd`-keys

jhillest, 2023-11-25:
You might find this guide useful:

[https://www.redhat.com/sysadmin/yaml-beginners](url)

markusressel, 2023-11-25:
@thyTwilightGoth when posting config file contents please use triple backticks (see markdown syntax), otherwise indents are hard to see.

About your problem:

1. Yes, fan2go shows errors when the yaml syntax is invalid. As you have seen in your second comment. There should be no need to use an external tool, but it doesnt hurt either.

2. The entries in the `fans:` section has two separate indentation levels. There is the first level, which has the same config options for all fan types, and the second level, which contains only fan type specific config options. You have only specified the second level but omitted the first one. Have a look at the example config. **Every** fan config entrx needs a `curve: mycurve` line.

markusressel, 2024-02-27:
Closing due to inactivity.

closed 2024-02-27 by markusressel