markusressel/fan2go #266: Misconfig? Fans ramp to minand then never up

jwefers, 2023-11-27:
**Describe the bug**
I set a config including a 'cpu' sensor and a 'gpu' sensor, reading from hwmon and nvidia-smi respectively. The sensors work fine apparently, `fan2go sensor -i cpu|gpu' spit out the expected temps, without any newline though.
```yaml
sensors:
  - id: cpu
    hwmon:
      platform: k10temp-pci-00c3
      index: 1
  - id: gpu
    cmd:
      exec: /usr/bin/nvidia-smi
      args: ["--query-gpu=temperature.gpu", "--format=csv,noheader"]
```

I set linear curves for two groups of fans, depending on these sensors:
```yaml
curves:
  - id: cpu_curve_sidefans
    linear:
      sensor: cpu
      steps:
        - 60000: 89 # k10temp outputs millidegrees
        - 95000: 255
  - id: gpu_curve_bottomfans
    linear:
      sensor: gpu
      steps:
        - 40: 0
        - 70: 128
```

and finally associated them to the fans:
```yaml
fans:
  - id: sidefans
    hwmon:
      platform: nct6792-isa-0290
      rpmChannel: 2
      pwmChannel: 2
    neverStop: true
    curve: cpu_curve_sidefans

  - id: bottomfans
    hwmon:
      platform: nct6792-isa-0290
      rpmChannel: 1
      pwmChannel: 1
    neverStop: true
    curve: gpu_curve_bottomfans
```

running as `sudo fan2go --verbose`, i can observe:
* bottomfans go slowly down to 0pwm, good
* sidefans go up to 106pwm, although the input sensor is below the min step (here: ~42000), which says pwm=89
* **and then...the fan speeds just stay there, never adjusting to rising temps**

**Expected behavior**
fans adhere to curves and temp changes

**Screenshots**
[image]

**Desktop (please complete the following information):**

- Distro: Arch Linux
- `uname -a`: Linux julian-endeavour-desktop 6.6.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 20 Nov 2023 23:18:21 +0000 x86_64 GNU/Linux

- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.8.1

**Additional context**
Add any other context about the problem here.

jwefers, 2023-11-27:
Currently debugging the code, and three observations so far:

**1)**
The code multiplies any given user temp values with 1000, assuming that all measurement values are millidegrees. That is not the case for nvidia-smi, which gives degrees. So i had to change my sensor to
```yaml
  - id: gpu
    cmd:
      exec: /usr/bin/bash
      args:
        - "-c"
        - 'echo "$(( $(/usr/bin/nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader) * 1000 ))"'
```
and my steps to full degrees. Contrary to what i was suggestest by looking at `fan2go detect`, which shows the raw millidegree values.

I'll hope i find time to contribute to the docs to highlight that.

**2)**
 There is a line "adjusting" the target PWM value: https://github.com/markusressel/fan2go/blob/b7cebcca5a80d9f701c451f33124a3ad45ff1bbd/internal/controller/controller.go#L426C45-L426C46
`target = minPwm + int((float64(target)/fans.MaxPwmValue)*(float64(maxPwm)-float64(minPwm)))`, which throws off the calculation.

To me, this says "scale the desired percentage of maxPwm to the configured pwm window". I think this is a bad idea, as you let the user explicitly state their desired step points (as i did). And the results leave the user-defined step boundaries.

**3)**
The PidLoop is overshooting. My bottom fan should not go over 128, as per curve. Yet:
```
  DEBUG   Setting Fan PWM of 'bottomfans' to 99 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 101 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 103 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 105 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 107 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 109 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 111 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 113 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 115 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 117 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 119 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 121 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 123 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 125 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 127 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 129 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 131 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 133 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 135 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 137 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 139 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 141 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 139 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 137 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 135 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 133 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 131 ...
  DEBUG   Setting Fan PWM of 'bottomfans' to 129 ...
```
I am looking at iterations of the PidLoop to understand what it does, but once at 129 (where it shouldn't be), it does not step down to 128 . indicating that there is an issue with small distances between current and desired.

I'll continue investigating, and already thanks for even writing this software. Abandoning Windows, i had to leave FanControl behind and your SW is the closest to it there is :)

jwefers, 2023-11-27:
Adding a debug line in LinearSpeedCurve.Evaluate, seeing idle behaviour:
```
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 108 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 106 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 104 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 102 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '41°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 100 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '40°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '38°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '40°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 98 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '40°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '40°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 96 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '40°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '40°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 94 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 92 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 90 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Setting Fan PWM of 'bottomfans' to 8 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 88 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Setting Fan PWM of 'bottomfans' to 6 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '34°'. Desired PWM: 89
[... cut ...]
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '39°'. Desired PWM: 0
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '41°'. Desired PWM: 4
```

jwefers, 2023-11-27:
To demonstrate that the "scaling" of 2) is wrong, here is a log with that line, resulting in speeds just go up:
```
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 3 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '40°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 89 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 5 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '39°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 91 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 7 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '37°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 93 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 9 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 95 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 11 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 97 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 13 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 99 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 15 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '46°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 101 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '46°'. Desired PWM: 26
  DEBUG   Setting Fan PWM of 'bottomfans' to 17 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '36°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 103 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 19 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 105 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 21 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 107 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 23 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 109 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 25 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 111 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 27 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '35°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
  DEBUG   Setting Fan PWM of 'sidefans' to 113 ...
  DEBUG   Evaluating curve 'gpu_curve_bottomfans' according to sensor 'gpu' temp '45°'. Desired PWM: 21
  DEBUG   Setting Fan PWM of 'bottomfans' to 29 ...
  DEBUG   Evaluating curve 'cpu_curve_sidefans' according to sensor 'cpu' temp '37°'. Desired PWM: 89
  DEBUG   Evaluating curve 'gpu_curve_sidefans' according to sensor 'gpu' temp '45°'. Desired PWM: 89
```

markusressel, 2024-02-26:
Hey @jwefers , sorry for not getting back to you sooner.

First of all: Thank you for the time and energy you already invested into this. There are definitely things about the current logic that are not optimal. I would also love to improve this, as the whole PID thing has caused issues and confusion in the past, but unfortunately I do not have the time to deep dive into this. I wasn't even able to read all of your comments yet, I will try to get a read this week.

About the linear approach: Yes, this is much simpler, and possibly the right approach, however we have to be careful about changing such integral behavior at this point, as it might break existing users fan control. Dealing with this probably requires implementing some kind of system to let the user choose between different approaches (pid vs. linear) to be able to default to the old one. This is the biggest reason why I wasn't able to "just merge" your PR.

When coming up with the PID loop logic, the main goal I had in mind was proving a flexible way to let the user specify how fan2go should approach its internally computed target pwm, with a default that provides a reasonably fast but also "smooth" behavior. I have since then had my doubts whether a PID loop was the right choice for this, especially since there are currently two separate PID loops acting at the same time. In hindsight, I think one of them is a bad idea (which might be what you are also saying here), but changing it requires a lot of care, which means time, which I didn't have (and still don't have much of). But maybe we can figure something out through a discussion here and within the PR :crossed_fingers: , so the first step is for me to read up on it.

There is a lot to unpack here so I will followup with a comment later, but feel free to share you thoughts in the meantime, if you want. Thx.

jwefers, 2024-02-26:
Hi Markus,

thanks for replying. I agree with everything you said, and in fact continued my PR _on my machine_, introducing the linear approach as a configurable alternative to the PID default.

Prob. same as you, life hit and i left it alone, did not get around to game anyway. Let me try to push my work-in-progress on this within the next days/weeks. At current speed, within a year i'll have a fully fleshed configurable new feature, that does not break any existing setups :laughing:

markusressel, 2024-02-29:
These comments are only based on your comments within this issue, I will have a look at your PR next.

> The code multiplies any given user temp values with 1000, assuming that all measurement values are millidegrees. 

Yes. This is also documented in the README. I can understand the confusion though, its not consistent everywhere. Changing this would also be a breaking change though, so I am not sure how we could improve this. Maybe adding an optional unit parameter?

> So i had to change my sensor to [millidegrees] and my steps to full degrees.

Same thing here, changing it would be a breaking change, but for the second we have at least full control, so adding a unit directly to the value would be an option (something like `1000ms` or `50°`/`50000m°`)

> ```target = minPwm + int((float64(target)/fans.MaxPwmValue)*(float64(maxPwm)-float64(minPwm)))```
> To me, this looks like an attempt to "scale the desired percentage of maxPwm to the configured pwm window". I think this is a bad idea, as you let the user explicitly state their desired step points (as i did), already including min and max values. And the results unfortunately leave the user-defined step boundaries, defying (my) user expectation.

I am not sure if the current implementation does what I want, but to explain what I tried to achieve with it:

All curves are expected to return a value in [0..255] when calling `Evaluate()` on them, as documented here: https://github.com/markusressel/fan2go/blob/e197de03718aa7465127abb612d0655cca7132af/internal/curves/curve.go#L13

Since the user has defined hard limits on min and max values for the same [0..255] range, this code is suppsed to map the [0..255] range the curve operates on to the [minPwm..maxPwm] range specified by the user for the relevant fan. This is supposed to ensure that the target pwm is always within the user defined [minPwm..maxPwm] bounds, but that obviously doesn't seem to be working correctly, and maybe the approach is not the best.

> The PidLoop is overshooting.

Thats not totally unexpected since thats how PID loops operate. Having pid loops for both the curve as well as the fan controller makes the whole system pretty hard to follow though, but I don't think many people are using a PID based curve either.

In the worst case, there are multiple things adding delay:
- sensor values are averaged
- the curve value is based on a PID loop
- the actual fan pwm applied is also based on a separate PID loop

All of them have the same goal: smoothing out the fan speed transition. This might be too much on too many levels. I am not sure what is best to keep though.

The sensor averaging is useful for removing random outliers, but since it can be smoothed out lateron too I think this can be removed.  
The curve PID is an optional feature, so while I wouldn't use it myself, it doesn't hurt the rest when using linear or functional curves.  
The fan controller PID is (I think) the one thats the important one to smooth out fan speed transitions, but even that one should be optional, since physics itself will also provide some amount of smoothing, which might be enough for some users. This is where your PR comes in with adding the option to change how the control loop works.

markusressel, 2024-09-24:
Just to give an update on this issue:

I have made a lot of adaptations and improvements, including a boatload of tests in #267 which will hopefully be merged relatively soon. This should fix the issues of the set boundaries beeing exceeded as well as the control logic failing to approach the target.

markusressel, 2025-05-04:
I have made further improvements on the PID loop control algorithm in #360 which takes sub integer changes into account properly. This should lead to the PWM value very slowly creeping towards the actual target value, even if the PID loop requests a PWM change of f.ex. `0.4`, which cannot be applied to the fan. Instead, these micro changes are summed up in the control algorithm loop until they are big enough to be able to apply them to the fan.

markusressel, 2025-05-08:
I will close this issue for now, if this is still a problem, let me know.
Thx.

closed 2025-05-08 by markusressel