markusressel/fan2go #273: Aquacomputer Octo can change platform name between reboots, device connections

A-t-B-o, 2024-02-25:
**Is your feature request related to a problem? Please describe.**
I have an Aquacomputer Octo in my rig that controls multiple fan-groups based on some sensors.
It works beautifully with fan2go.
However, it's platform-name changes sometimes.
When that happens, the fans are stuck at maximum RPM until I edit the fan2go YAML config file with the new platform-name 
(for example change it from "octo-hid-3-6" to "octo-hid-3-8") and restart the fan2go service demon.

The full physical path for the device is below:
>/sys/class/hwmon/hwmon6$ readlink -f ./device
/sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:69:00.0/usb5/5-10/5-10:1.1/0003:0C70:F011.0006

(That last digit can change from time to time causing the issue)

**Describe the solution you'd like**
It would be very nice if we could use a wildcard or regex at the platform-name in the YAML file to permanently resolve this issue.
Something like "octo-hid-3-*" or Something like "octo-hid-3-?"

For example in the YAML file we could use:
- id fans_in
  hwmon:
    platform: octo-hid-3-?
    rpmChannel: 2
...

**Describe alternatives you've considered**
I didnt find any other way to solve this. Every other fan control service has the same issue with the octo controllers.

**Additional context**
screenshot attached about the fan2go detect
[image]

markusressel, 2024-02-26:
Hi @A-t-B-o 

this should already be possible, as stated in the description here:
https://github.com/markusressel/fan2go?tab=readme-ov-file#hwmon-1

Is this not working as expected?

A-t-B-o, 2024-02-26:
OMG Im blind, sorry Markus!  All is good, sorry for the bother.

closed 2024-02-26 by A-t-B-o