markusressel/fan2go #283: info provided by detect conflicts with readme

ShalokShalom, 2024-03-29:
**Describe the bug**
Hi there

I am about to configure fan2go for the first time, and this is what I see: 

[image]

And this is what the README says: 

[image]

This conflicts in at least two ways. 
First, you provide only the (unreliable?) index number, and second, I dont see a RPM/PWM channel at all.

markusressel, 2024-03-30:
Hey @ShalokShalom , thx for the report! :heart: 

Indeed you are correct. The support for specifying the pwm channel more precisely was added in #206 . The reason this is not the default is that we wanted to avoid a breaking change. In the comments of that PR the need for also implementing this for sensors was also mentioned:

> It occurs to me a similar change is probably warranted for the temperature sensors. Probably worth a separate PR.

This simply hasn't happened yet, so it is understandably confusing. If you would like to give it a try implementing this I am happy to work with you on a PR!

ShalokShalom, 2024-03-30:
Hello :)

I would prefer to just understand, what I am supposed to do right now 👍🏻 

I then can adjust the README, so that it makes sense.

markusressel, 2024-03-30:
You don't see a pwm/rpm channel because you only see sensors. Do you expect to see fan pwm/rpm values? If so, can you see them in the output of `sensors`?

As long as using the index does not cause any problems you can use the index shown in the fan2go detect output. I am not sure what the exact question is 🤔

ShalokShalom, 2024-03-31:
Ah, ok. Maybe let us adjust the README then?

markusressel, 2024-04-05:
Sorry, but I am still not sure what part of the documentation is unclear and how exactly you would want to update/modify it. Feel free to open a PR with your proposed changes, so we can discuss it in more detail.

Since the original question is answered I will close this issue for now, but I will also create a new issue for implementing the same "channel" logic for sensors. Thx!

closed 2024-04-05 by markusressel

ShalokShalom, 2024-04-05:
This output of detect shows way more information, and you literally recommend to use values, that simply dont exist in the real app. 

https://github.com/markusressel/fan2go?tab=readme-ov-file#fans

markusressel, 2024-04-06:
`fan2go detect` shows fans and sensors in separate tables. The linked output has both, while your output only contains sensors. Sensors don't currently have a channel in the table, fans do.

> You don't see a pwm/rpm channel because you only see sensors. Do you expect to see fan pwm/rpm values? If so, can you see them in the output of `sensors`?

The section in the README you linked to is called "Fans", which is not coincidence. Its not about sensors. It is correct, that we should also implement something similar for sensors, which is why I have creadet #286 , but (as I mentioned) this hasn't happened yet, so it only applies to fans right now.

If you have a suggestion on how to improve the README, feel free to open a PR with an improved version of the README, I am more than happy to discuss a possible improvement with you.