markusressel/fan2go #298: PWM values occasionally get reset to 0

electricboy6, 2024-06-13:
**Describe the bug**
On my computer, the PWM values for my fans and AIO pump occasionally don't set or reset to zero. This is usually a nonissue, but my computer just hard reset due to thermal throttling. The CPU usually never goes above ~75C, so it is not a cooling issue. The logs indicate that all of the fans just stopped ("PWM mode stuck to 0"), then I have nothing more as my computer hard shut down.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
The fans do not randomly stop/change speed

**Screenshots**
```-- Boot 721f1644f6a54f99af4660f3bc3eb4a7 --
Jun 11 19:16:47 (REDACTED) systemd[1]: Started Advanced Fan Control program.
Jun 11 19:16:47 (REDACTED) fan2go[1476]: fan2go
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Gathering sensor data for aioPump...
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Gathering sensor data for cpuFans...
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Gathering sensor data for intake...
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Gathering sensor data for exhaust...
Jun 11 19:16:47 (REDACTED) fan2go[1476]: INFO: Gathering sensor data for gpuBottomIntake...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Loading fan curve data for fan 'aioPump'...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Loading fan curve data for fan 'intake'...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Loading fan curve data for fan 'cpuFans'...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Loading fan curve data for fan 'gpuBottomIntake'...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Loading fan curve data for fan 'exhaust'...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Using pwm map override from config...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: PWM settings of fan 'aioPump': Min 0, Start 50, Max 255
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Starting controller loop for fan 'aioPump'
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Using pwm map override from config...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: PWM settings of fan 'intake': Min 0, Start 50, Max 255
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Starting controller loop for fan 'intake'
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Using pwm map override from config...
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: PWM settings of fan 'cpuFans': Min 0, Start 50, Max 255
Jun 11 19:16:49 (REDACTED) fan2go[1476]: INFO: Starting controller loop for fan 'cpuFans'
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: Using pwm map override from config...
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: PWM settings of fan 'exhaust': Min 0, Start 50, Max 255
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: Starting controller loop for fan 'exhaust'
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: Using pwm map override from config...
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: PWM settings of fan 'gpuBottomIntake': Min 0, Start 50, Max 255
Jun 11 19:16:50 (REDACTED) fan2go[1476]: INFO: Starting controller loop for fan 'gpuBottomIntake'
...
Jun 13 13:31:00 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:31:00 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:31:13 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:31:13 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:31:13 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:31:15 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:31:15 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:31:15 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:31:18 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:31:42 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:31:47 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:31:52 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:31:52 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:31:52 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:31:57 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:32:02 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'cpuFans' to "1": PWM mode stuck to 0
Jun 13 13:32:05 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'cpuFans' to "1": PWM mode stuck to 0
Jun 13 13:32:08 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'cpuFans' to "1": PWM mode stuck to 0
Jun 13 13:32:21 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:32:23 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:32:23 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:32:23 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:32:24 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:32:26 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:32:27 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
[... 74 lines cut ...]
Jun 13 13:36:21 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:36:25 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:36:26 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:36:26 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:36:26 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:36:28 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:36:30 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:36:31 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:36:31 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
Jun 13 13:36:36 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'gpuBottomIntake' to "1": PWM mode stuck to 0
Jun 13 13:36:41 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'intake' to "1": PWM mode stuck to 0
Jun 13 13:36:41 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:36:41 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'exhaust' to "1": PWM mode stuck to 0
Jun 13 13:36:41 (REDACTED) fan2go[1476]: ERROR: Unable to set Fan Mode of 'aioPump' to "1": PWM mode stuck to 0
```

**Desktop (please complete the following information):**

- Distro: Linux Mint 21.3 with 6.5 kernel series
- `uname -a`: `Linux (REDACTED) 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0 with libsensors version 3.6.0`
- `fan2go version`: `dev` (it was just built from source. `diff` indicates that it is the exact same as 0.8.1)

**Additional context**
I have been having errors like this for a long time, but it never actually caused an issue with my fans. This was the first time that something bad actually happened due to errors like this.

markusressel, 2024-06-17:
Hi @electricboy6 thx for reporting :heart: 

Without a way to reproduce your issue it will be difficult to find a solution. How sure are you that this is caused by fan2go and not by the hardware driver? Changing the fan mode is done by simply writing a number to a file, if that fails the linux system is preventing fan2go from changing the mode (and also anyone else). Knowing the reason why it is preventing this change would be necessary. Maybe fan2go breaks the driver somehow, maybe the driver has a bug, or maybe its something else entirely.

electricboy6, 2024-06-18:
Unfortunately, I have no way of knowing what it is caused by. At the very least, I hoped that posting an issue here would allow me to get some suggestions on help for it. If it helps at all, I'm running an Asus X570 board (I know the other info, but I doubt it would be more useful) with a 5900x. I gained the ability to control my fans when I updated to the 6.2 kernel series in case that helps at all.

It seems that it is likely that it may be the driver, but I have no way of knowing how unless you have an idea on how to catch it. It happens often enough that it would be trivial to find an occurrence to troubleshoot further if you know how. Sorry for wasting your time if it isn't fan2go, but it would be nice to know at the very least why it is happening or how to fix it.

closed 2025-02-01 by electricboy6