markusressel/fan2go #300: fan2go can not set fan speed

recallmenot, 2024-07-01:
fan2go is unable to set the fan speed on my system

fancontrol and coolercontrol can do so just fine, enabling the corresponding services one-at-a-time.
But when I do `sudo systemctl start fan2go`, I get the log below from `journalctl -u fan2go -f`, repeating of course unless I stop the service.
I can also set values manually (as root) in /sys/class/hwmon/hwmon3/ by echoing to the pwm# files and the values stay how I set them. The fans respond accordingly.
Somehow just fan2go seems to be unable to set fan speed values.

OS: Manjaro latest
kernel: Linux 6.10.0-rc4-1-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Jun 18 11:21:09 UTC 2024 x86_64 GNU/Linux
sensors: sensors version 3.6.0+git with libsensors version 3.6.0+git
fan2go: 0.8.1 (fan2go-git from AUR)
mainboard: MSI PRO Z690-A
module: [nct6687d](https://github.com/Fred78290/nct6687d)

journalctl -u fan2go -f
```
fan2go[3095]: INFO: Gathering sensor data for front_high...
fan2go[3095]: INFO: Loading fan curve data for fan 'front_low'...
fan2go[3095]: INFO: Loading fan curve data for fan 'front_high'...
fan2go[3095]: INFO: Loading fan curve data for fan 'cpu'...
fan2go[3095]: INFO: Loading fan curve data for fan 'back'...
fan2go[3095]: INFO: FanController: Using saved value for pwm map of Fan 'front_low'
fan2go[3095]: INFO: PWM settings of fan 'front_low': Min 180, Start 200, Max 255
fan2go[3095]: INFO: Starting controller loop for fan 'front_low'
fan2go[3095]: INFO: FanController: Using saved value for pwm map of Fan 'cpu'
fan2go[3095]: INFO: PWM settings of fan 'cpu': Min 0, Start 0, Max 129
fan2go[3095]: INFO: Starting controller loop for fan 'cpu'
fan2go[3095]: INFO: FanController: Using saved value for pwm map of Fan 'front_high'
fan2go[3095]: INFO: PWM settings of fan 'front_high': Min 0, Start 0, Max 0
fan2go[3095]: INFO: Starting controller loop for fan 'front_high'
fan2go[3095]: WARNING: Fan 'back' has not yet been analyzed, starting initialization sequence...
fan2go[3095]: INFO: FanController: Using saved value for pwm map of Fan 'back'
fan2go[3095]: INFO: Measuring RPM curve...
fan2go[3095]: WARNING: WARNING: Increasing minPWM of front_low from 0 to 1, which is supposed to never stop, but RPM is 0
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 33
fan2go[3095]: WARNING: WARNING: Increasing minPWM of cpu from 0 to 1, which is supposed to never stop, but RPM is 0
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 33
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 33
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 33
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 0
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 0
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 0
fan2go[3095]: WARNING: PWM of cpu was changed by third party! Last set PWM value was: 58 but is now: 0
```

fan2go detect
```
> coretemp-isa-0000
  Sensors  Index  Label                       Value
           1      Package id 0 (temp1_input)  29000
           2      Core 0 (temp2_input)        26000
           3      Core 4 (temp6_input)        29000
           4      Core 8 (temp10_input)       26000
           5      Core 12 (temp14_input)      26000
           6      Core 16 (temp18_input)      28000
           7      Core 20 (temp22_input)      28000
           8      Core 28 (temp30_input)      29000
           9      Core 29 (temp31_input)      29000
           10     Core 30 (temp32_input)      29000
           11     Core 31 (temp33_input)      29000

> nvme-pci-0100
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  38850
           2      Sensor 1 (temp2_input)   38850
           3      Sensor 2 (temp3_input)   47850

> nct6687-isa-0a20
  Fans     Index  Channel  Label          RPM  PWM  Auto 
           1      1        CPU Fan        0    33   false
           2      2        Pump Fan       980  255  false
           3      3        System Fan #1  0    153  false
           4      4        System Fan #2  703  153  false
           5      5        System Fan #3  619  153  false
           6      6        System Fan #4  0    153  false
           7      7        System Fan #5  0    153  false
           8      8        System Fan #6  0    153  false
  Sensors  Index  Label                     Value
           1      CPU (temp1_input)         29000
           2      System (temp2_input)      38500
           3      VRM MOS (temp3_input)     32000
           4      PCH (temp4_input)         58000
           5      CPU Socket (temp5_input)  32500
           6      PCIe x1 (temp6_input)     36000
           7      M2_1 (temp7_input)        31500

> nvme-pci-0500
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  37850
           2      Sensor 1 (temp2_input)   37850
           3      Sensor 2 (temp3_input)   45850
```

/etc/fan2go/fan2go.yaml (validates fine):
```
fans:
  # A user defined ID.
  # Used for logging only
  - id: cpu
    hwmon:
      platform: nct6687
      rpmChannel: 1
      pwmChannel: 1
    neverStop: true
    curve: cpu_curve
  - id: back
    hwmon:
      platform: nct6687
      rpmChannel: 2
      pwmChannel: 2
    neverStop: true
    curve: cpu_curve
  - id: front_low
    hwmon:
      platform: nct6687
      rpmChannel: 3
      pwmChannel: 3
    neverStop: true
    curve: cpu_curve
    minPwm: 180
    maxPwm: 255
    startPwm: 200
  - id: front_high
    hwmon:
      platform: nct6687
      rpmChannel: 4
      pwmChannel: 4
    neverStop: true
    curve: cpu_curve
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: coretemp
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  - id: cpu_curve
    # The type of the curve, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Sensor input value at which the curve is at minimum speed
      min: 25
      # Sensor input value at which the curve is at maximum speed
      max: 60
```

markusressel, 2024-10-09:
@recallmenot Hey there, sorry for the late response.

There have been lots of changes for 0.9.0, can you please retest on the latest version?

Also:
```
 PWM settings of fan 'front_high': Min 0, Start 0, Max 0
```
That doesn't sound like a valid configuration, did the initialization fail?
Did you try overriding these values via the config?

edt-xx, 2024-10-20:
I've got something similar happening:

```
Oct 20 09:56:19 grover fan2go[937]: INFO: PWM settings of fan 'top': Min 95, Start 105, Max 255
Oct 20 09:56:19 grover fan2go[937]: INFO: Starting controller loop for fan 'top'
Oct 20 09:56:19 grover fan2go[937]: INFO: Computing pwm map...
Oct 20 09:56:21 grover fan2go[937]: INFO: PWM settings of fan 'front1': Min 50, Start 85, Max 255
Oct 20 09:56:21 grover fan2go[937]: INFO: Starting controller loop for fan 'front1'
Oct 20 09:56:21 grover fan2go[937]: INFO: Computing pwm map...
Oct 20 09:56:21 grover fan2go[937]: WARNING: WARNING: Increasing minPWM of top from 0 to 1, which is supposed to never stop, but RPM is 0
```
Should fan2go not set the minPWM back to the 95 I have in the config?  (this is with fan2go-git built 20 Oct).

markusressel, 2025-03-16:
Hey @edt-xx , the log in this case is misleading. The `0` and `1` are "offsets" that are added to the original `minPwm` value.
I will create a small PR to cleanup the message. I am still not sure about the origin of the original problem.