markusressel/fan2go #310: `getRpm` required when using `fan` `cmd`

diggit, 2024-09-20:
**Describe the bug**
In readme, there is an example with cmd controlled fan and `getPwm` is marked as optional. I also found #265 where the `getPwm` is also omitted and it was not an error. my configuration does not define it, but validation complains

**To Reproduce**
``` yaml
fans:
  - id: main
    curve: cpu_curve
    cmd:
      setPwm:
        exec: /usr/bin/ectool
        args: [ "fanduty", "0", "%pwm%" ]

      # (optional) Command to retrieve the current PWM value (0..255)
      # getPwm:
      #   exec: /usr/bin/nvidia-settings
      #   args: [ "-a", "someargument" ]
      # # (optional) Command to retrieve the current RPM value
      getRpm:
        # wrapped ectool output because it returns "Fan 0 RPM: 1234" to get only "1234"
        exec: /usr/bin/wrap_ectool_getfanrpm
        args: []

    minPwm: 10
    startPwm: 15
    maxPwm: 10
```

Validation
```
fan2go config validate
INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
ERROR   Validation failed: fan main: missing getPwm configuration
```

**Expected behavior**
Validation passes

**Desktop (please complete the following information):**

- Distro: Arch Linux
- `uname -a`: `Linux xlylt 6.10.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 12 Sep 2024 17:21:02 +0000 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0+git with libsensors version 3.6.0+git`
- `fan2go version`: `0.8.1`

markusressel, 2024-09-20:
Hi @diggit , thx for reporting.

I am afraid this documentation has been wrong for a long time already. The `getPwm` property is indeed **not optional**. Reading the current PWM value is used in a lot of places in the internal workins of fan2go, so its currently not feasible to simply omit it. The same is true for the `setPwm` property. Only the "RPM sensor" is really optional.

I will update the README to reflect this.

closed 2024-09-20 by markusressel

diggit, 2024-09-20:
On my machine (Framework 13 AMD), `ectool` (EC control utility) is not able to read out fan PWM duty, as EC does not support it.
My current workaround is to wrap the the `ectool` PWM set calls with shell script which also stores the value in file in tmp. getPwm then only read that value.
IMO such fallback could be also implemented in fan2go (with come warning).

wrap_ectool_fanduty_set
``` sh
#!/bin/sh

FILE="/tmp/wrap_pwmduty_cached"

# Check if exactly one argument is provided
if [ "$#" -ne 1 ]; then
	echo "wrapper expects exactly one argument" >&2
	exit 1
fi

# Write the argument to the file
echo "$1" > $FILE

ectool fanduty $1
```

wrap_ectool_fanduty_get
``` sh
#!/bin/sh

FILE="/tmp/wrap_pwmduty_cached"

if [ -f "$FILE" ]; then
	cat "$FILE"
else
	echo 0
fi
```

PR #312 (2024-09-20): fix wrong documentation
fixes #310