markusressel/fan2go #317: Error reading PWM value of fan file_fan: file is empty

ettom, 2024-10-02:
**Describe the bug**
In about ~1 minute intervals, the following error is reported:
`WARNING  Error reading PWM value of fan file_fan: file is empty: /tmp/fan2go/pwm`

**To Reproduce**
Config file:
```
curves:
- id: cpu_curve
  linear:
    max: 80
    min: 40
    sensor: cpu_package
dbPath: /tmp/fan2go/db
fans:
- curve: cpu_curve
  file:
    path: /tmp/fan2go/pwm
    rpmPath: /tmp/fan2go/fan_input
  id: file_fan
sensors:
- hwmon:
    index: 1
    platform: coretemp
  id: cpu_package
```
```
mkdir -p /tmp/fan2go
echo 0 > /tmp/fan2go/fan_input
echo 0 > /tmp/fan2go/pwm
fan2go -c config.yml
```
...and observe errors popping up.

**Expected behavior**
No error messages/log pollution.

**Desktop (please complete the following information):**

- Distro: NixOS
- `uname -a`: `Linux 6.6.52 #1-NixOS SMP PREEMPT_DYNAMIC Wed Sep 18 17:24:10 UTC 2024 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0 with libsensors version 3.6.0`
- `fan2go version`: 0.9.0

**Additional context**
Some unrelated observations and ideas:
The README says that to print fan curve data (RPM/PWM curve), do `sudo fan2go curve`. This doesn't seem to work and just prints the help text for curve related commands. I guess the correct command would be `sudo fan2go fan curve -i <fan_id>`?.

Doing `fan2go -c config.yml fan init -i file_fan` with the same config results in a panic:
```
 INFO  Using configuration file at: config.yml
 INFO  Using persistence at: /tmp/fan2go/db
  FATAL   Failed to create fan controller for fan 'file_fan': Curve with ID 'cpu_curve' not found
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
	/home/runner/go/pkg/mod/github.com/pterm/pterm@v0.12.79/prefix_printer.go:351
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0x1228b80, {0xc7caa6, 0x4a}, {0xc00004fb80, 0x2, 0x2})
	/home/runner/go/pkg/mod/github.com/pterm/pterm@v0.12.79/prefix_printer.go:293 +0x1e6
github.com/markusressel/fan2go/internal/ui.Fatal({0xc7caa6, 0x4a}, {0xc00004fb80, 0x2, 0x2})
	/home/runner/work/fan2go/fan2go/internal/ui/logging.go:67 +0x8a
github.com/markusressel/fan2go/internal/controller.NewFanController({0xd80070, 0xc00004a2b0}, {0xd84900, 0xc0000d6120}, {0xd75300, 0xc00007e140}, 0xbebc200)
	/home/runner/work/fan2go/fan2go/internal/controller/controller.go:89 +0x157
github.com/markusressel/fan2go/cmd/fan.init.func2(0xc00024c600?, {0xc000233440?, 0x4?, 0xc4f874?})
	/home/runner/work/fan2go/fan2go/cmd/fan/init.go:31 +0x172
github.com/spf13/cobra.(*Command).execute(0x122d6a0, {0xc000233400, 0x4, 0x4})
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.1/command.go:985 +0xaaa
github.com/spf13/cobra.(*Command).ExecuteC(0x122bfa0)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/markusressel/fan2go/cmd.Execute()
	/home/runner/work/fan2go/fan2go/cmd/root.go:89 +0xf0
main.main()
	/home/runner/work/fan2go/fan2go/main.go:25 +0xf
```

I also found it confusing that the logs don't seem to contain anything about the initialization procedure. When deleting the database and restarting fan2go, I'd expect some log messages about starting/finishing the initialization procedure.

markusressel, 2024-10-03:
Hi @ettom thx for the report :+1: 

> Expected behavior
> No error messages/log pollution.

For that to happen we would need to know what its caused by. Is there an issue with the code in fan2go that reads the file?

> Please consider writing to the PWM file atomically. The current implementation just does a plain old write, which inevitably results in a failing read (by whatever is reading the PWM file) at some point. Writing to a temporary file and then renaming it to the target file is atomic on Linux.

I am not aware of this, can you elaborate and/or link some sources? Could this be the source of the error messages?
If this is something that can be implemented without causing side effects, feel free to open a PR :heart: 

> The README says that to print fan curve data (RPM/PWM curve), do sudo fan2go curve. This doesn't seem to work and just prints the help text for curve related commands. I guess the correct command would be sudo fan2go fan curve -i <fan_id>?.

The README is wrong indeed, your suggestion is correct. I have updated it. The CLI is not perfect yet regarding this though as there is currently no way to list all fan curve data because the "fan id" parameter is required.

> Doing fan2go -c config.yml fan init -i file_fan with the same config results in a panic

Thx for the report, this seems to have been introduced recently, I am working on a fix.

> I also found it confusing that the logs don't seem to contain anything about the initialization procedure. When deleting the database and restarting fan2go, I'd expect some log messages about starting/finishing the initialization procedure.

The initialization is currently only done for hwmon fans. This is not by design though, I just have not been using anything except hwmon devices myself so I have limited testing on everything else. If you would like to add support for other types as well, feel free to open a PR :+1:

ettom, 2024-10-03:
>For that to happen we would need to know what its caused by. Is there an issue with the code in fan2go that reads the file?

I'm not sure. My best guess is that maybe it's caused by fan2go itself writing to the file from another thread, which causes the reading to fail from time to time. 

>I am not aware of this, can you elaborate and/or link some sources? Could this be the source of the error messages?
If this is something that can be implemented without causing side effects, feel free to open a PR ❤️

So if my guess is correct, then the error messages are indeed caused by the same effect, although I was talking about the RPM file, not the PWM file.

The issue is that reading a file while another process is writing to it is not a reliable method to exchange data. The reading process might do the read just after the file was truncated but before new data was written. A way around this is to use locking (inefficient, especilly at the rates fan2go operates) or the rename trick. A few relevant links:
https://stackoverflow.com/questions/29261648/atomic-writing-to-file-on-linux
https://stackoverflow.com/questions/7054844/is-rename-atomic

Sorry, but I currently don't have the time to open any PRs.

markusressel, 2024-10-07:
Did you use fan2go-tui at the same time in another shell, or just the fan2go daemon?

fan2go-tui is prone to causing exceptions due to race conditions and I am aware of that for a while already, but it's simply not that easy for me to fix.

If this happens even without fan2go-tui we should be able to write a test that reproduces your issue. Can you come up with a testcase?

ettom, 2024-10-08:
I did not use fan2go-tui. The initial report contains the exact steps to reproduce it. Not sure about the testcase, sorry.

markusressel, 2024-10-08:
@ettom I have added atomic file writes to the file fan code.
Let me know if the issue is resolved.

ettom, 2024-10-08:
Seems to be working great! Thanks!

closed 2024-10-08 by ettom