markusressel/fan2go #320: fan2go-0.9.0 test failures

stkw0, 2024-10-07:
**Describe the bug**
Tests fails to run. The more easy to fix tests fails due to echo command being located in /bin/echo in at least some systems instead of /usr/bin/echo

**To Reproduce**
go test -v ./...

**Desktop (please complete the following information):**

- Distro: [e.g. Arch Linux]: Gentoo
- `uname -a`: Linux 6.11.0-gentoo-x86_64
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: 0.9.0

**Additional context**
Logs of tests that fails:
```
=== RUN   TestCmdFan_GetRpm
    cmd_test.go:146:
        	Error Trace:	/var/tmp/portage/sys-power/fan2go-0.9.0/work/fan2go-0.9.0/internal/fans/cmd_test.go:146
        	Error:      	Received unexpected error:
        	            	cannot execute /usr/bin/echo: lstat /usr/bin/echo: no such file or directory
        	Test:       	TestCmdFan_GetRpm
    cmd_test.go:147:
        	Error Trace:	/var/tmp/portage/sys-power/fan2go-0.9.0/work/fan2go-0.9.0/internal/fans/cmd_test.go:147
        	Error:      	Not equal:
        	            	expected: 1000
        	            	actual  : 0
        	Test:       	TestCmdFan_GetRpm
```
(there are 2 other failures related with echo command)

```
=== RUN   TestPidCurveAboveTarget
    pid_test.go:419:
        	Error Trace:	/var/tmp/portage/sys-power/fan2go-0.9.0/work/fan2go-0.9.0/internal/curves/pid_test.go:419
        	Error:      	Not equal:
        	            	expected: 22
        	            	actual  : 23
        	Test:       	TestPidCurveAboveTarget
        	Messages:   	loop: 4
--- FAIL: TestPidCurveAboveTarget (1.21s)
=== RUN   TestPidCurveWayAboveTarget
    pid_test.go:453:
        	Error Trace:	/var/tmp/portage/sys-power/fan2go-0.9.0/work/fan2go-0.9.0/internal/curves/pid_test.go:453
        	Error:      	Not equal:
        	            	expected: 45
        	            	actual  : 46
        	Test:       	TestPidCurveWayAboveTarget
        	Messages:   	loop: 4
--- FAIL: TestPidCurveWayAboveTarget (1.21s)
```

markusressel, 2024-10-07:
Hi @stkw0 :nerd_face: 

I am a bit confused, why would tests need to run on a variety of systems?

stkw0, 2024-10-07:
As part of the standard practices when packaging software, tests are executed on the OS or distro to ensure that the software works correctly before making it available to users through the official repositories. If a package is found to not work properly, the problematic version is delayed, discarded or patched downstream. If a program doesn't support a given OS or distro it's usually not packaged.

It's not that tests should run on a variety of systems, but on the systems that fab2go would like to support. In this case, I assumed that Linux users in general were the target audience, including Gentoo users and users of derivate distros.

If there are particular Linux distros that fan2go decide not to support, I think will be clearer to add that information to the README, saving bot, potential issues reports for unsupported distros and effort from maintainers from packaging an unsupported software.

markusressel, 2024-10-07:
@stkw0 thx for the clarification!

I am happy to include support for other distros besides arch, but it has to come from the community. This is a single person project and I use arch as my main distro, so that's what I can support by myself. I do not want to exclude other distros, but I also can't add support for them and not even packages for them.

If there is a way to support both distros, feel free to open a PR for it.
The tests are still suboptimal in some places anyway though, since they (only some) depend on the execution speed because I didn't know how to mock the time yet. That's probably something we should fix as well since tests will be flaky otherwise.

I appreciate your interest in my project ❤️

stkw0, 2024-10-07:
Yeah that sounds good. If there is interest in supporting other distros I am happy with sending a PR. Hopefully will also help other distros to package fan2go :)

closed 2024-10-08 by markusressel