markusressel/fan2go #326: maxPwmChangePerCycle buggy, turns fans off

jwefers, 2024-11-20:
Thanks for picking up my PR and making the direct algo happen. I think i dis

**Describe the bug**
when using controlAlgorithm.direct.maxPwmChangePerCycle with any value, it wants to set the correct pwm value, but then sets the fan speed to zero instead.

As last time, i'll try to properly debug this when i find time, but right now, Stalker 2 was more important.

**To Reproduce**
Steps to reproduce the behavior:

config:
```yaml
fans:
  - id: myfan
    hwmon:
      platform: nct6792-isa-0290
      rpmChannel: 3
      pwmChannel: 3
    neverStop: true
    curve: topfans_curve
    controlAlgorithm:
        direct:
            maxPwmChangePerCycle: 10
    pwmMap:
      0: 0
      255: 255
```
I assume curve and sensor do not matter, use anything.

What happens is, Debug shows correct desired PWM, but actually set PWM is zero and fan stays off.
When just using direct without maxPwmChangePerCycle, it works as expected.

**Expected behavior**
As temp in curves are changing, target PWM is slowly approached

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- Distro: Arch
- `uname -a`: Linux xxx 6.11.9-arch1-1 1 SMP PREEMPT_DYNAMIC Sun, 17 Nov 2024 16:06:17 +0000 x86_64 GNU/Linux
- `sensors -v`:  sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.9.0

markusressel, 2024-11-21:
Hi @jwefers , thx for reporting.

I am not sure what was going on with this code, I might have let copilot go a bit far without thinking about it :smile: 
Please have a look at the PR I have created, it simplifies the logic greatly and should fix the issue. I am unsure about the "timedelta" component though, so input from a user that actually has this usecase would be great.

https://github.com/markusressel/fan2go/pull/327#discussion_r1852946168

Thx!

markusressel, 2024-11-28:
As mentioned in #327 I have merged the pending PR.
@jwefers Please let me know if the implementation makes sense to you.

jwefers, 2024-11-30:
i'll get to it eventually, its not urgent - i'll get around to investigate/test the fix

jwefers, 2024-12-21:
I found my bug in the config - i wanted to circumvent your pwm mapping, so i used the pwmMap with only two targets. Little did i know that those are not the ONLY possible values. I thought they are linear steps, just like a linear curve. So the nearest pwn target was always either 0 or 255.

BTW, about your `// TODO: figure out how to normalize the change to the timedelta`:

```patch
diff --git a/internal/control_loop/direct.go b/internal/control_loop/direct.go
index 1850c0a..4098b0e 100644
--- a/internal/control_loop/direct.go
+++ b/internal/control_loop/direct.go
@@ -31,13 +31,13 @@ func NewDirectControlLoop(
 func (l *DirectControlLoop) Cycle(target int, current int) int {
 	loopTime := time.Now()
 
-	//dt := loopTime.Sub(l.lastTime).Seconds()
+	dt := int(loopTime.Sub(l.lastTime).Seconds())
 
 	l.lastTime = loopTime
 
 	var stepTarget = float64(target)
 	if l.maxPwmChangePerCycle != nil {
-		maxChangeValue := *l.maxPwmChangePerCycle
+		maxChangeValue := *l.maxPwmChangePerCycle * dt
 		// TODO: figure out how to normalize the change to the timedelta
 		//stepTarget = float64(*l.maxPwmChangePerCycle) * dt
 

```

closed 2024-12-21 by jwefers

jwefers, 2024-12-21:
oh right, `master` works - the version from AUR with the same config does not...not sure what the diff is

markusressel, 2024-12-21:
Thx for the update, yes the `pwmMap` causes quite a bit of confusion, but I still think its a good solution for the problem it was created for. It just seems to me that its hard for people to understand what this problem is and if they need to use `pwmMap` to solve their issue. Not sure how to improve that besides better documentation.

As for your suggestion to normalize the value: I thought the same, but this does't work, because if the cycle is longer than 1 second it would lead to a change that is bigger than the specified `maxPwmChangePerCycle`. I am not even sure if its a good idea to normalize this anyway, since its a "per cycle" config and the cycle time is configurable. I had a lot of headaches thinking about this and gave up at some point, therefore the TODO.

The AUR _is_`master`. The package checks out the latest commit of `master`, builds it and "installs" it. There is no diff.

jwefers, 2024-12-21:
Better docs might be fine. I understand the problem but imho the user should just be made aware where their fans start and make them regard that fact manually.

Yeah, the value should be called `maxPwmChangePerSecond`, thats what the change would do. If you want per configurable cycle, its fine as is i think.

Weird, i could swear the AUR version behaved differently that then cloned version...will check another time. Guess for now at least my personal issue is solved and i can spend my rare spare time with stalker

jwefers, 2024-12-21:
thanks for making this piece of SW :)