markusressel/fan2go #330: Incorrect file location in zsh autocompletion setup command on Manjaro

tarlacac, 2024-11-30:
**Describe the bug**
The command to set up fan2go zsh autocompletion on Linux for every new session does not result in the correct file location on Manjaro. The instructed command from `fan2go completion zsh --help` is `fan2go completion zsh > "${fpath[1]}/_fan2go"`.

**To Reproduce**
Steps to reproduce the behavior:

1. On Manjaro, in a terminal, run `fan2go completion zsh --help` to get the instructions.
2. Following the instructions for Linux, run `fan2go completion zsh > "${fpath[1]}/_fan2go"`
3. See error:
```zsh
% fan2go completion zsh > "${fpath[1]}/_fan2go"
zsh: no such file or directory: /usr/local/share/zsh/site-functions/_fan2go
```
4. Verify `/usr/local/share/zsh` does not exist:
```zsh
% ls /usr/local/share/zsh
ls: cannot access '/usr/local/share/zsh': No such file or directory
```
5. Verify `/usr/share/zsh` exists:
```zsh
% ls /usr/share/zsh
5.9/	    manjaro-zsh-config	p10k-portable.zsh  plugins/  site-functions/
functions/  manjaro-zsh-prompt	p10k.zsh	   scripts/  zsh-maia-prompt
```

**Expected behavior**
My zsh site-functions directory is `/usr/share/zsh/site-functions`, so I would expect the recommended command to use that directory. Alternatively, I would expect a note warning that the location might be `/usr/share/zsh/site-functions` instead of `/usr/local/share/zsh/site-functions`.

**Desktop (please complete the following information):**

- Distro: Manjaro
- `uname -a`: `Linux <hostname> 6.6.54-2-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Oct  8 03:11:08 UTC 2024 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0+git with libsensors version 3.6.0+git`
- `fan2go version`: `0.8.1`

**Additional context**
I'm using zsh 5.9. As far as I know, I have not done any configuration to zsh except a few `p10k configure` calls for the theme. From some brief searching, it appears that zsh using either `/usr/share` or `/usr/local/share` varies based on Linux distro.

markusressel, 2024-11-30:
Hi @tarlacac , thx for the report :+1: 

The `completion` command is part of the [cobra](https://github.com/spf13/cobra) golang library for building CLI interfaces and not built or maintained by me. If the location is different on manjaro, its probably because the manjaro devs thought it would be a good idea to diverge from the arch default for some reason. You can let the cobra maintainers know on their GitHub (mentioned above) and they might do something about it. Please make sure to search for existing issues first before creating one. If you do, please link the issue here so I can update the library to include their fix, if a PR is merged on their end.

Unfortunately thats all I am able to do from this end.

tarlacac, 2024-12-01:
@markusressel No worries, thanks for the info! I'll check out the issues over at cobra and link back if I find/create something relevant.

FYI I'm not super familiar with PKGBUILD files, but from a glance I believe the standard Arch Linux `extra/zsh` and `extra/zsh-completions` packages use `/usr/share`, not `/usr/local/share`, so I don't think Manjaro is going against the default here. It does seem like Oh My Zsh considers the non-`*/local/*` directories [to be insecure](https://github.com/ohmyzsh/ohmyzsh/blob/master/lib/compfix.zsh) in some permissions/ownership setups, so I can see an argument to avoid those locations, though.

tarlacac, 2024-12-01:
I did some digging and couldn't find anything about this particular issue in the cobra repo, so I submitted an issue there: https://github.com/spf13/cobra/issues/2204. I'll ping you if there's a PR merged for it!

markusressel, 2025-05-08:
Hey @tarlacac , since there was a lot of silence around this issue, I will close it for now.
If a new `cobra` version is released, I will include it swiftly.
Until then, there is nothing I can do about it.
Thx :)

closed 2025-05-08 by markusressel