markusressel/fan2go #338: Fails with write only pwm file

symphorien, 2025-01-27:
**Describe the bug**
on my system the fan speed can be set but not read:

```
echo 255 > /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon6/pwm1
# works, fan goes full speed
cat /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon6/pwm1
cat: /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon6/pwm1: Invalid argument
```

fan2go seems to not like this situation at all.

**To Reproduce**
Steps to reproduce the behavior:

write this configuration file:
```
fans:
  - id: cpu_fan
    hwmon:
      platform: dell_smm-isa-0000
      rpmChannel: 1
    curve: cpu_curve
sensors:
  - id: cpu_sensor
    hwmon:
      platform: dell_smm-isa-0000
      index: 1
curves:
  - id: cpu_curve
    linear:
      sensor: cpu_sensor
      steps:
        - 50: 0
        - 60: 70
        - 70: 128
        - 80: 255
```

run fan2go:
```
#  fan2go -c fan2go.yaml
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Version: dev
 INFO  Using configuration file at: fan2go.yaml
 WARNING  Cannot read pwm value of cpu_fan
 INFO  Gathering sensor data for cpu_fan...
 INFO  Loading fan curve data for fan 'cpu_fan'...
 WARNING  Fan 'cpu_fan' has not yet been analyzed, starting initialization sequence...
 INFO  Computing pwm map...
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
 WARNING  Error reading PWM value of fan cpu_fan: read /sys/class/hwmon/hwmon6/pwm1: invalid argument
[... 244 lines cut ...]
  ERROR   Error sending notification: exit status 1
panic: read /sys/class/hwmon/hwmon6/pwm1: invalid argument

goroutine 165 [running]:
github.com/markusressel/fan2go/internal.RunDaemon.func7()
	github.com/markusressel/fan2go/internal/backend.go:139 +0x16b
github.com/oklog/run.(*Group).Run.func1({0xc00014a0c0?, 0xc000158570?})
	github.com/oklog/run@v1.1.0/group.go:38 +0x29
created by github.com/oklog/run.(*Group).Run in goroutine 1
	github.com/oklog/run@v1.1.0/group.go:37 +0x5a
```
fan2go exits after about a second.

**Expected behavior**
fan2go only writes to /sys/class/hwmon/hwmon6/pwm1 depending on sensor readings

**Desktop (please complete the following information):**

- Distro: NixOS 24.11
- `uname -a`: `Linux uqbar 6.12.9 #1-NixOS SMP PREEMPT_DYNAMIC Thu Jan  9 12:33:55 UTC 2025 x86_64 GNU/Linux`
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: 0.9.0

**Additional context**
this is distinct from https://github.com/markusressel/fan2go/issues/229 On my system pwm_enable does not even exist, and writing pwm1 works without any preparation.

symphorien, 2025-01-27:
Note that current fan speed is available as:
```
/sys/devices/platform/dell_smm_hwmon/hwmon/hwmon6/fan1_input
3624
```
this is what the `sensors` command displays.

markusressel, 2025-03-16:
Hi @symphorien , thx for the report.

This is currently expected. The algorithms used and the sanity checks that are done to be able to warn the user about precarious situations require both read and write access to the "pwm file" of a fan.

I guess it would be possible to add a "per fan" configuration option (something like `disablePwmSanityChecks`) to disable safety checks and just assume that setting a pwm value works (if there is no error) without checking the value afterwards. Though this would have obvious drawbacks, it would at least be possible to control fans like these at all. Using the RPM value as an alternative is sadly not useful.

markusressel, 2025-03-18:
@symphorien I have come up with a possible solution for your situation. Would you be able to test the referenced branch on your machine to report back if it woks as intended?

markusressel, 2025-03-20:
I have merged the proposed changes and have tested them on my system using a `file` based fan.
Let me know if it works for you.

symphorien, 2025-03-24:
Hi,

Reading the diff of the PR you merged I could not determine how to adapt my configuration file to use the new feature you implemented.

I tried to rerun the reproduction steps with the exact same configuration file as before and it still attempts to read the pwm file, even though it does detect that it is not possible:
```
$  sudo ./fan2go --config conf.yaml 
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Version: dev
 INFO  Using configuration file at: conf.yaml
 INFO  Gathering sensor data for cpu_fan...
 INFO  Loading fan curve data for fan 'cpu_fan'...
 WARNING  Fan 'cpu_fan' has not yet been analyzed, starting initialization sequence...
 INFO  Computing pwm map...
 WARNING  Fan 'cpu_fan' does not support PWM sensor, using default PWM map
 INFO  Measuring RPM curve...
  ERROR   Fan cpu_fan: Unable to measure current PWM
 INFO  Fan controller for fan cpu_fan stopped.
  ERROR   Error sending notification: exit status 1
panic: read /sys/class/hwmon/hwmon7/pwm1: invalid argument

goroutine 167 [running]:
github.com/markusressel/fan2go/internal.RunDaemon.func7()
	/home/symphorien/src/fan2go/internal/backend.go:144 +0x154
github.com/oklog/run.(*Group).Run.func1({0xc0002ae140?, 0xc000286630?})
	/home/symphorien/go/pkg/mod/github.com/oklog/run@v1.1.0/group.go:38 +0x29
created by github.com/oklog/run.(*Group).Run in goroutine 1
	/home/symphorien/go/pkg/mod/github.com/oklog/run@v1.1.0/group.go:37 +0x5a
```

markusressel, 2025-03-24:
Hey @symphorien , thx for getting back! 😄 

> Reading the diff of the PR you merged I could not determine how to adapt my configuration file to use the new feature you implemented.

This is correct, I attempted to avoid having to set anything in the config at all.
There was a case during fan initialization I have missed, and pushed b8c249c3 to hopefully fix it.
Please try again with the latest version.

symphorien, 2025-03-24:
thank you it works fine!

markusressel, 2025-03-24:
The value of a curve is not applied directly, instead a `ControlLoop` determines how to apply the value determined by a curve. This is done to provide smooth ramping up and down, and prevent instantanious full speed on short load spikes. There are two different control loop algorithms to choose from, if you don't like the default, you can switch to the "direct" one.

See: https://github.com/markusressel/fan2go?tab=readme-ov-file#control-algorithms

The value should approach 255 over time though, if thats not the case, you might have uncovered another bug. If thats the case, please open a new issue for that :)

PR #346 (2025-03-20): FeaturePwmSensor FeatureFlag
The new FeatureFlag `FeaturePwmSensor` is used to avoid reading pwm value if the fan does not support it.
FIxes #338