markusressel/fan2go #356: sensors show spd5118-i2c-10-51 and spd5118-i2c-10-53, fan2go detect reports as spd5118 for both

edt-xx, 2025-04-26:
**Describe the bug**
sensors show:

spd5118-i2c-10-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +38.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

spd5118-i2c-10-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +37.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

fan2go detect reports:

> spd5118
  Sensors  Index  Label                        Value
           1      hwmon10/temp1 (temp1_input)  38750

> spd5118
  Sensors  Index  Label                        Value
           1      hwmon11/temp1 (temp1_input)  37750

since both hwmod10 and 11 both have the same name in fan2go they cannot be selected correctly

**To Reproduce**
Steps to reproduce the behavior:

1. lmsensors-detect to find a spd5118
2. sensors 
3. fan2go detect
4. See error

**Expected behavior**
fan2go detect should show the correct name as found in sensors output so both can be selected

**Desktop (please complete the following information):**

- Distro: arch
- `uname -a`: Linux grover 6.14.4-1-stable-git #1 SMP PREEMPT_DYNAMIC Sat, 26 Apr 2025 00:26:44 +0000 x86_64 GNU/Linux

- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: dev (fan2go-git 0.9.0.r94.g097910d-1 built 26 Apr 2025)

**Additional context**
seems to be the same type of issue as found in #191

markusressel, 2025-04-26:
Hi @edt-xx , thx for the report.

Should be fixed now, please try with latest version and report back.

closed 2025-04-26 by markusressel

edt-xx, 2025-04-26:
Looks good now.  Thanks

markusressel, 2025-04-26:
@edt-xx Thx, should be fixed as well now.

edt-xx, 2025-04-27:
Confirmed.  Thanks again!

PR #357 (2025-04-26): fix i2c identifier
fixes #356