markusressel/fan2go #358: init broken? sets rpm values during init sequence within milliseconds

jwefers, 2025-05-03:
**Describe the bug**
New hardware; deleting db file, having adapted config file; running `sudo fan2go --verbose`, which, being the first run, triggers the init sequence and generates a PWM map.

But it sets the rpm values for both my fans milliseconds apart; first it goes through setting all possible rpm values (taking 2 seconds to go from 0 to 255); then waits for settling. The resulting PWM map is obviously completely broken.

**To Reproduce**
Steps to reproduce the behavior: see 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- Distro: [e.g. Arch Linux]
- `uname -a`:
- `sensors -v`:
- `fan2go version`:

**Additional context**
Log of `sudo fan2go --verbose` with no prior db file:
```
 INFO  Verbose output enabled
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Version: 0.9.0
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for cpufan...
 INFO  Gathering sensor data for topfans...
 INFO  Loading fan curve data for fan 'topfans'...
 INFO  Loading fan curve data for fan 'cpufan'...
 WARNING  Fan 'topfans' has not yet been analyzed, starting initialization sequence...
 INFO  Computing pwm map...
  DEBUG   Setting Fan PWM of 'topfans' to 255 ...
  DEBUG   Setting Fan PWM of 'topfans' to 254 ...
  DEBUG   Setting Fan PWM of 'topfans' to 253 ...
  DEBUG   Setting Fan PWM of 'topfans' to 252 ...
  DEBUG   Setting Fan PWM of 'topfans' to 251 ...
 WARNING  Fan 'cpufan' has not yet been analyzed, starting initialization sequence...
 INFO  Computing pwm map...
  DEBUG   Setting Fan PWM of 'cpufan' to 255 ...
  DEBUG   Setting Fan PWM of 'topfans' to 250 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 254 ...
  DEBUG   Setting Fan PWM of 'topfans' to 249 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 253 ...
  DEBUG   Setting Fan PWM of 'topfans' to 248 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 252 ...
  DEBUG   Setting Fan PWM of 'topfans' to 247 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 251 ...
  DEBUG   Setting Fan PWM of 'topfans' to 246 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 250 ...
  DEBUG   Setting Fan PWM of 'topfans' to 245 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 249 ...
  DEBUG   Setting Fan PWM of 'topfans' to 244 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 248 ...
  DEBUG   Setting Fan PWM of 'topfans' to 243 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 247 ...
  DEBUG   Setting Fan PWM of 'topfans' to 242 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 246 ...
  DEBUG   Setting Fan PWM of 'topfans' to 241 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 245 ...
  DEBUG   Setting Fan PWM of 'topfans' to 240 ...
  DEBUG   Setting Fan PWM of 'cpufan' to 244 ...
  DEBUG   Setting Fan PWM of 'topfans' to 239 ...
[... 563 lines cut ...]
  DEBUG   PWM map of fan 'cpufan': map[0:32 1:32 2:32 3:32 4:32 5:32 6:32 7:32 8:32 9:32 10:32 11:32 12:32 13:32 14:32 15:32 16:32 17:32 18:32 19:32 20:32 21:32 22:32 23:32 24:32 25:32 26:32 27:32 28:32 29:32 30:32 31:72 32:72 33:72 34:72 35:72 36:72 37:72 38:72 39:72 40:72 41:72 42:72 43:72 44:72 45:72 46:72 47:72 48:72 49:72 50:72 51:72 52:72 53:72 54:72 55:72 56:72 57:72 58:72 59:72 60:72 61:72 62:72 63:72 64:72 65:72 66:72 67:72 68:72 69:72 70:72 71:72 72:112 73:112 74:112 75:112 76:112 77:112 78:112 79:112 80:112 81:112 82:112 83:112 84:112 85:112 86:112 87:112 88:112 89:112 90:112 91:112 92:112 93:112 94:112 95:112 96:112 97:112 98:112 99:112 100:112 101:112 102:112 103:112 104:112 105:112 106:112 107:112 108:112 109:112 110:112 111:112 112:154 113:154 114:154 115:154 116:154 117:154 118:154 119:154 120:154 121:154 122:154 123:154 124:154 125:154 126:154 127:154 128:154 129:154 130:154 131:154 132:154 133:154 134:154 135:154 136:154 137:154 138:154 139:154 140:154 141:154 142:154 143:154 144:154 145:154 146:154 147:154 148:154 149:154 150:154 151:154 152:154 153:194 154:194 155:194 156:194 157:194 158:194 159:194 160:194 161:194 162:194 163:194 164:194 165:194 166:194 167:194 168:194 169:194 170:194 171:194 172:194 173:194 174:194 175:194 176:194 177:194 178:194 179:194 180:194 181:194 182:194 183:194 184:194 185:194 186:194 187:194 188:194 189:194 190:194 191:194 192:194 193:194 194:234 195:234 196:234 197:234 198:234 199:234 200:234 201:234 202:234 203:234 204:234 205:234 206:234 207:234 208:234 209:234 210:234 211:234 212:234 213:234 214:234 215:234 216:234 217:234 218:234 219:234 220:234 221:234 222:234 223:234 224:234 225:234 226:234 227:234 228:234 229:234 230:234 231:234 232:234 233:234 234:255 235:255 236:255 237:255 238:255 239:255 240:255 241:255 242:255 243:255 244:255 245:255 246:255 247:255 248:255 249:255 250:255 251:255 252:255 253:255 254:255 255:255]
 INFO  PWM settings of fan 'cpufan': Min 0, Start 0, Max 0
 INFO  Starting controller loop for fan 'cpufan'
  DEBUG   Evaluating curve 'cpu_curve_for_top'. Sensor 'cpu' temp '45°'. Desired PWM: 100
  DEBUG   Evaluating curve 'gpu_curve'. Sensor 'gpu' temp '36°'. Desired PWM: 100
  DEBUG   Evaluating curve 'topfans_curve'. Curve values: '[100 100]' Desired PWM: 100
  DEBUG   Setting PWM of topfans to 0, found closest distinct PWM value at 0, applying PWM Map yields 255
  DEBUG   Evaluating curve 'cpu_curve'. Sensor 'cpu' temp '45°'. Desired PWM: 25
 WARNING  PWM of cpufan was changed by third party! Last set PWM value was: 255 but is now: 234
  DEBUG   Setting PWM of cpufan to 0, found closest distinct PWM value at 0, applying PWM Map yields 32
  DEBUG   Setting Fan PWM of 'cpufan' to 32 ...
  DEBUG   Evaluating curve 'cpu_curve_for_top'. Sensor 'cpu' temp '45°'. Desired PWM: 100
  DEBUG   Evaluating curve 'gpu_curve'. Sensor 'gpu' temp '36°'. Desired PWM: 100
  DEBUG   Evaluating curve 'topfans_curve'. Curve values: '[100 100]' Desired PWM: 100
  DEBUG   Setting PWM of topfans to 0, found closest distinct PWM value at 0, applying PWM Map yields 255
  DEBUG   Evaluating curve 'cpu_curve'. Sensor 'cpu' temp '44°'. Desired PWM: 25
  DEBUG   Setting PWM of cpufan to 0, found closest distinct PWM value at 0, applying PWM Map yields 32
  DEBUG   Evaluating curve 'cpu_curve_for_top'. Sensor 'cpu' temp '44°'. Desired PWM: 100
  DEBUG   Evaluating curve 'gpu_curve'. Sensor 'gpu' temp '36°'. Desired PWM: 100
  DEBUG   Evaluating curve 'topfans_curve'. Curve values: '[100 100]' Desired PWM: 100
  DEBUG   Setting PWM of topfans to 0, found closest distinct PWM value at 0, applying PWM Map yields 255
  DEBUG   Evaluating curve 'cpu_curve'. Sensor 'cpu' temp '44°'. Desired PWM: 25
  DEBUG   Setting PWM of cpufan to 0, found closest distinct PWM value at 0, applying PWM Map yields 32
^C
```

markusressel, 2025-05-04:
Hi @jwefers ,

I am not sure I understand the problem 🤔 
Initialization of fans happens concurrently, if enabled in the config. So the fact that you are seeing both fans beeing analyzed at the same time is fine - if it is enabled in the config.

But I think you also have an issue with the timing of the analysis. I guess I could figure it out manually using the log data you provided, but to make sure we are on the same page, can you try to explain it in your words first, please?

markusressel, 2025-05-04:
> It sets the rpm values for both my fans milliseconds apart; first it goes through setting all possible rpm values (taking 2 seconds to go from 0 to 255); then waits for settling. The resulting PWM map is obviously broken.

Setting all **pwm** values (I assume thats what you meant) is only done to figure out which pwm values can actually be set on the device. The part of the initialization that analyzes the fan curve (pwm -> rpm map) follows.

The method that tries to set all pwm values does so very quickly, to avoid a long delay on startup, as this check is (currently) done on each startup for each fan, unless its set manually in the config. The time it takes to do this is currently hardcoded to 5 milliseconds between each pwm value check. It would be possible to make this configurable, if thats what your issue is about. You could also set this manually in the config and skip the automatic detection entirely.

jwefers, 2025-05-05:
Good morning,
i understand, this is not desired pwm -> reported pwm, but just testing if a pwm value can be set at all?

My worries stems from the resulting pwm map:

```
 DEBUG   Distinct PWM value targets of fan topfans: [0]
  DEBUG   PWM map of fan 'topfans': map[0:255 1:255 2:255 3:255 4:255 5:255 6:255 7:255 8:255 9:255 10:255 11:255 12:255 13:255 14:255 15:255 16:255 17:255 18:255 19:255 20:255 21:255 22:255 23:255 24:255 25:255 26:255 27:255 28:255 29:255 30:255 31:255 32:255 33:255 34:255 35:255 36:255 37:255 38:255 39:255 40:255 41:255 42:255 43:255 44:255 45:255 46:255 47:255 48:255 49:255 50:255 51:255 52:255 53:255 54:255 55:255 56:255 57:255 58:255 59:255 60:255 61:255 62:255 63:255 64:255 65:255 66:255 67:255 68:255 69:255 70:255 71:255 72:255 73:255 74:255 75:255 76:255 77:255 78:255 79:255 80:255 81:255 82:255 83:255 84:255 85:255 86:255 87:255 88:255 89:255 90:255 91:255 92:255 93:255 94:255 95:255 96:255 97:255 98:255 99:255 100:255 101:255 102:255 103:255 104:255 105:255 106:255 107:255 108:255 109:255 110:255 111:255 112:255 113:255 114:255 115:255 116:255 117:255 118:255 119:255 120:255 121:255 122:255 123:255 124:255 125:255 126:255 127:255 128:255 129:255 130:255 131:255 132:255 133:255 134:255 135:255 136:255 137:255 138:255 139:255 140:255 141:255 142:255 143:255 144:255 145:255 146:255 147:255 148:255 149:255 150:255 151:255 152:255 153:255 154:255 155:255 156:255 157:255 158:255 159:255 160:255 161:255 162:255 163:255 164:255 165:255 166:255 167:255 168:255 169:255 170:255 171:255 172:255 173:255 174:255 175:255 176:255 177:255 178:255 179:255 180:255 181:255 182:255 183:255 184:255 185:255 186:255 187:255 188:255 189:255 190:255 191:255 192:255 193:255 194:255 195:255 196:255 197:255 198:255 199:255 200:255 201:255 202:255 203:255 204:255 205:255 206:255 207:255 208:255 209:255 210:255 211:255 212:255 213:255 214:255 215:255 216:255 217:255 218:255 219:255 220:255 221:255 222:255 223:255 224:255 225:255 226:255 227:255 228:255 229:255 230:255 231:255 232:255 233:255 234:255 235:255 236:255 237:255 238:255 239:255 240:255 241:255 242:255 243:255 244:255 245:255 246:255 247:255 248:255 249:255 250:255 251:255 252:255 253:255 254:255 255:255]
```

so after 2 seconds this init sequence takes, with an empty DB, this is the resulting map. All values point to 255.

markusressel, 2025-05-05:
> i understand, this is not desired pwm -> reported pwm, but just testing if a pwm value can be set at all?

Correct. There are fans that only support f. .ex. 0, 127 and 255. This "sweep" over the 0-255 range is trying to detect that.

> My worries stems from the resulting pwm map

That definitely doesn't look right. If your fan (controller) supports the full 0-255 range, maybe it is a little slow to react?
Can you test if manual changes to the pwm value stick and how long it takes for a new value to be reported back when read out again? You can use `fan2go fan --id mydan speed 255` and `fan2go fan --id mydan speed` to check, but writing and reading the PRM file directly also might help in debugging.

jwefers, 2025-05-05:
okay it takes around a second for an rpm write to reflect back on read

markusressel, 2025-05-05:
Please also take a look at #324 where a similar thing was discussed. Adding an option to bypass the logic is an option, but it would still require people to find out about the problem, set the config option, and hope for the best.

I would like to enhance the user experience by automatically applying different defaults based on the detected controller, but this requires some research to be able to detect controller hardware (vendor, model, etc) as well as a system that allows to specify custom configuration options per such hardware configuration.

If this were to be implemented, your controller would "just work" and use this "bypassPwmMap" option by default.

jwefers, 2025-05-05:
I will write great documentation to the README, the best documentation.

edt-xx, 2025-05-09:
I've also noticed problems with init.  If you have a fan that has a controller loop started and you run init on it (eg. a physical fan has been replaced etc) the results are very very unpredictable and you often see messages that some third party (init vs control loop) has set the PWM value.  I suspect that init is not checking correctly if a control loop is active for the fan.  I can get a good curve by using reset and restarting fan2go with any minpwm/maxpwm/startpwm commented out.

edt-xx, 2025-05-09:
That is not at all obvious from help... I suggest either updating init to complain if a controller loop is active for the fan (or fans) requested or stop the controller loop(s) for the fan (or fans) requested by init.  It would be nice to be able to use init with an active daemon after replacing a fan - though reset and restart also works.   Even if nothing is done an update to the docs/help would be really nice.

markusressel, 2025-05-09:
@edt-xx feel free to create a PR 👍

markusressel, 2025-05-24:
@jwefers I have put some thought into the whole `pwmMap` thing again and I think there is still an issue with how its currently used.

Your thought that:

> this is not desired pwm -> reported pwm, but just testing if a pwm value can be set at all?

Was actually hinting at that, as this is the way the `pwmMap` is currently automatically populated when it is not specified manually. The issue is, that this mapping is then used when applying a curve value (which is always in [0..255] range) to the fan range (which might be in some other range), which doesn't always make sense.

F.ex.: When I set a PWM of `5` to my GPU, it will report back with a PWM value of `6`. Don't ask me why, it just does. The automatically detected `pwmMap` would detect this and store `5 -> 6` in it. Now when the fan controller gets a curve value of `5`, it wants to set a PWM value of `5` to the fan, but the `pwmMap` is used and maps it to `6` instead. However, applying the map in that case is actually wrong, because setting value of `6` will result in a reported PWM value of `7`! which results in the "third party" detection springing into action on every single controller loop, as fan2go successfully confused itself.

So... I think there is a mixup of how the `pwmMap` is computed and how its used/applied in the controller logic. The computation actually creates the "desired pwm -> reported pwm" map you mentioned, but thats not what is actually needed, at least not for controlling the fan. We don't really care what PWM value is reported by the fan, we only care about our internal value range, which is always in `[0..255]`. It might be missing some values, and it might be capped at the top and bottom, but the scale is always `[0..255]`.

The current automatic computation algorithm can still be used for detecting values that yield the same result (maybe setting `4` and setting `5` both yield a reported value of `6`).

So the easy fix would be to simply not apply the `pwmMap` when controlling the fan. This would prevent the issue explained in the example above. **However** the `pwmMap` is also currently already used by people to change what values are actually set by the fan controller.

This is a lot to take in, **so to summarize**...

I think there are two different use-cases that we need to cover that cannot both make use of the `pwmMap` alone in the same way to work.

1. We want to be able to change what value `X` is written to the PWM of a fan, based on a map. F.ex. if a fan only supports setting `[1,2,3]`, we need to map the `[0..255]` range to `[1,2,3]` somehow.
2. We want to know what PWM value `Y` is reported by the fan after writing a PWM value `X`. F.ex. setting a value of `5` might yield `6` after reading. **Crucially this not the same as in the first use-case**, as in that case setting a value of `255` might yield a value of `3`, but it could also be that setting a value of `3` yields a value of `3`, and everything above just doesn't work (throws error or whatnot). Yet, for mapping our internal range to the supported range of the fan, we cannot only use `1 -> 1, 2 -> 2, 3 -> 3` as this would result in all curve values >= 3 leading to full fan power, which is obviously not the desired behavior.

So... before we can think about expanding the configuration of the `pwmMap` config option, we need a clear and concise definition of what its supposed to do and be used for, then we need to implement a distinction for these two use-cases and provide a migration path (if necessary). You can still work on a PR to enhance the configuration now, but we should wait for this issue to be resolved first before merging it (and possibly making a migration much harder for existing users).

As this issue somewhat is beyond the scope of this issue, I will create a new one in a hot second.

markusressel, 2025-05-28:
@jwefers 
#371 is merged, the problem mentioned above should be resolved now.
This caused some conflicts in your PR but that was unavoidable.
Let me know if you need assistance.

DanielGibson, 2025-05-31:
> We want to know what PWM value Y is reported by the fan after writing a PWM value X. F.ex. setting a value of 5 might yield 6 after reading. 

Why do we want to know this? What is it good for?

So your GPU rounds the value you set up (or maybe in other cases down), because it might internally use a smaller range than 0..255 (maybe 0..100) - why does this matter?
If it matters at all, then for validation - but to do that IMHO you don't need a map, but just the maximum deviation between set and read value. Then when validating you check whether `abs(setPwm - gotPwm) <= maxDeviation` and that's it.

I think for this a map (or similar) would only make sense if the PWM values you read are entirely different from the ones you write (e.g. you write the raw value 128 and read value 3, or vice versa). But I doubt this actually happens anywhere?

.. except when the map has been created incorrectly, like it was the case in @jwefers's original post, or also with my nvidia backend.
Don't know if here we have a similar problem, but with nvidia/nvml, you can set a speed (between 0 and 100) and you can read a speed (also between 0 and 100).
But the value you read is not (immediately) the one you set - instead it represents the **actual, _current_** fan speed in percent.
This means after setting it you must wait for the fan to spin up/down accordingly, and when measuring it, it might be off by a point or two due to physics.

So if you want to create this map at all (or measure the max deviation), then it should best be done in the **second `fan init` step**, which already waits for the fans to spin up.

> We want to be able to change what value X is written to the PWM of a fan, based on a map. F.ex. if a fan only supports setting [1,2,3], we need to map the [0..255] range to [1,2,3] somehow.

Wouldn't in this case MinPWM be 1 and MaxPWM 3 and that's it? The values from the fan curve are already mapped to the range from MinPWM to MaxPWM (probably with simple math) so it should just work, without any map?

Or are there actually fans where you can only set e.g. 1, 5, 9 ?
In that case the map makes sense, of course - but probably only as a config option, not as something that is measured automagically

> #371 is merged, the problem mentioned above should be resolved now.

Do I have that fix in the nvml branch? Because there the second `fan init` step (apparently) didn't work because the first step created a completely wrong map (due to not waiting for the fan to spin up)

markusressel, 2025-06-02:
> If it matters at all, then for validation

Correct, to detect if fan2go is not the only one controlling the fan. This happens more often than one would think.

> But I doubt this actually happens anywhere?

You would be surprised 😄 

> But the value you read is not (immediately) the one you set - instead it represents the actual, current fan speed in percent.

Yes, this breaks the check fan2go is doing, so it should be turned off in that case, because it is impossible to make it work. Waiting for the value to settle is way to much of a headache to implement (and maintain) something around that. This was not possible to do until very recently, see #374 .

Since the goal of fan2go is to work out-of-the-box for new users, it would be great to be able to provide such settings as defaults if the hardware that is in use requires it. Such a system does - at the time of writing - not exist.

> In that case the map makes sense, of course - but probably only as a config option, not as something that is measured automagically

This is true, and I have come to the same conclusion when I recently redid the map creation logic in #372 . The pwmMap now only uses the setPwmToSetPwm map to determine which PWM values can be set without failure, and assumes a 1:1 relation. If this is not what is actually required, the user has to tell fan2go via the config. The logic that was used previously came to existence due to a mixup of different concepts in my head when I wrote it (because I don't actually have any use for it, it is all just for other users).

> Here's the output of fan init for one of my GPU fans

From what I can tell its working "as expected". The automatic `pwmMap` creation logic tries to only use pwm values that actually result in a different pwm value readout (thats what the "uniqueness" is about), but since Nvidia decided to return dynamically computed values (idk how people come up with this madness) this logic cannot work as intended. Again this is something a user would have to tell fan2go (since the driver doesn't), preferrably once in the fan2go repo and then its solved for all users.

markusressel, 2025-06-04:
> Again this is something a user would have to tell fan2go (since the driver doesn't), preferrably once in the fan2go repo and then its solved for all users.

This has now been "solved" by always reporting the `FeaturePwmSensor` flag as `false`, meaning "not supported". This results in fan2go not trying to analyze the pwm behavior and assuming a linear 1:1 relation instead, which is good enough for the tests that have been done up until this day.