markusressel/fan2go #364: panic: non-positive interval for NewTicker

ButteredCats, 2025-05-06:
**Describe the bug**
fan2go starts up, claims a few of my fans have not yet been analyzed (even though they should be), and then crashes after a few minutes.

**To Reproduce**
Run fan2go on the newest commit

**Expected behavior**
fan2go starts up like normal without trying to recalibrate fans and then controls them instead of crashing

**Screenshots**
[image]

**Desktop:**
- Distro: Arch Linux
- `uname -a`: 6.14.5-2-cachyos
- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.9.1

**Additional context**
I was previously running commit `54d420d` and this broke just today when I updated to the latest commit (`e9e001b`)

I've had fan2go setup for a while now without issues.
I've tried deleting the database and starting fan2go again but it didn't help.

markusressel, 2025-05-07:
Whoops, should be fixed in 907572cf .
Thx for reporting!

ButteredCats, 2025-05-07:
Works perfectly now, thanks for the quick fix!

closed 2025-05-07 by ButteredCats