markusressel/fan2go #366: function maximum not always working

edt-xx, 2025-05-08:
**Describe the bug**
my gpu_curve is set to be the maximum of 6 other curves.  they have values, but the gpu_curve value is zero

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
gpu_curve should be the maximum of the six included curges

**Screenshots**
see below

**Desktop (please complete the following information):**

- Distro: Arch

- `uname -a`
  Linux grover 6.15.0-rc5-1-stable-git #1 SMP PREEMPT_DYNAMIC Mon, 05 May 2025 13:31:39 +0000 x86_64 GNU/Linux

- `sensors -v`:
sensors version 3.6.0+git with libsensors version 3.6.0+git

- `fan2go version`:
dev (0.91 git including off by one fix)

**Additional context**

copied from http://localhost:9001/curve.  There is another function using maximum that includes all of the values below too.  it works. 

  "gpu_curve": {
    "config": {
      "id": "gpu_curve",
      "function": {
        "type": "maximum",
        "curves": [
          "edge_curve",
          "junc_curve",
          "mem_curve",
          "nvme0_curve",
          "nvme1_curve",
          "nvme2_curve"
        ]
      }
    },
    "value": 0
  },
  "junc_curve": {
    "config": {
      "id": "junc_curve",
      "linear": {
        "sensor": "junc",
        "min": 30,
        "max": 90,
        "steps": null
      }
    },
    "value": 36
  },
  "mem_curve": {
    "config": {
      "id": "mem_curve",
      "linear": {
        "sensor": "mem",
        "min": 30,
        "max": 88,
        "steps": null
      }
    },
    "value": 61
  },
  "nvme0_curve": {
    "config": {
      "id": "nvme0_curve",
      "linear": {
        "sensor": "nvme0",
        "min": 35,
        "max": 75,
        "steps": null
      }
    },
    "value": 88
  },
  "nvme1_curve": {
    "config": {
      "id": "nvme1_curve",
      "linear": {
        "sensor": "nvme1",
        "min": 35,
        "max": 85,
        "steps": null
      }
    },
    "value": 40
  },
  "nvme2_curve": {
    "config": {
      "id": "nvme2_curve",
      "linear": {
        "sensor": "nvme2",
        "min": 35,
        "max": 85,
        "steps": null
      }
    },
    "value": 40
  },

edt-xx, 2025-05-08:
Its seems to make a difference if I use gpu_curve in my all_curve instead of including the sensors in gpu_curve in all_curve.

markusressel, 2025-05-08:
Hi @edt-xx , 

1. Whats your config?
2. Do you see any errors in the log?

edt-xx, 2025-05-09:
The logs look clean, My working config is below (gpu_curve included in all_curve instead of the values from gpu_curve)

[fan2go.txt](https://github.com/user-attachments/files/20126766/fan2go.txt)

markusressel, 2025-05-15:
Thx for providing the config. At first glance I don't see an issue with it. 🤔 

> Its seems to make a difference if I use gpu_curve in my all_curve instead of including the sensors in gpu_curve in all_curve.

I am not sure I understand. Can you please try to explain a little more what you mean by this using example curve config sections?

Also: How did you verify that the value is zero, only through the API? Is the curve value really at zero, meaning also the fans that use this curve are at 0?

markusressel, 2026-02-21:
There hasn't been any activity here, and I have looked at the code but cannot find a reason to get a 0 value. Besides that there have been a lot of changes to the underlying control logic, so it might be fixed already.

I will close this issue, but feel free to reopen if it still exists. Thx!

closed 2026-02-21 by markusressel