markusressel/fan2go #369: version command always returns "dev"

marktuk, 2025-05-23:
**Describe the bug**
I was hoping to use the version command as part of a dockerfile but it only ever seems to return "dev" rather than a version number that maps to the releases in GitHub.

**To Reproduce**
Steps to reproduce the behavior:

1. Exectute the command `fan2go version`
4. Observe that the returned value is always "dev"

**Expected behavior**
The command returns the currently installed version number for fan2go

marktuk, 2025-05-23:
I installed it using the latest release:

```
# Install fan2go
RUN curl -L -o fan2go https://github.com/markusressel/fan2go/releases/latest/download/fan2go-linux-amd64 && \
    chmod +x fan2go && \
    cp ./fan2go /usr/bin/fan2go
```

markusressel, 2025-05-23:
Seems like the GitHub action doesn't use the `make` script to build the binaries, I will take a look.

markusressel, 2025-05-23:
Phew, that took longer than anticipated. New releases should now print the correct version.

markusressel, 2025-05-24:
@marktuk do you need a new release with this fix to continue?

markusressel, 2025-05-24:
@marktuk I have just released [0.9.2](https://github.com/markusressel/fan2go/releases/tag/0.9.2)
Let me know if it works now. Thx!

marktuk, 2025-05-25:
I'm still getting version returning "dev" in my docker container even after rebuilding it. Let me try building it on a different machine and I'll come back to you.

markusressel, 2025-05-25:
Works for me:
[image]