markusressel/fan2go #376: Issues in Documentation

DanielGibson, 2025-05-31:
Based on https://github.com/markusressel/fan2go/pull/373#issuecomment-2918033447

(might be that some of these are actual issues in the *implementation* as well)

* [ ] HwMonController has both Name and Platform, fan2go detect prints the `Name`, but in the config the "platform" is supposed to be used
 - [ ] `hwmon detect` does not mention "platform" anywhere, but I already changed this in #373
* [ ] the `fan2go detect` example shows `nct6798` - but in reality it looks like `nct6798-isa-0290`
* [ ] In `fan2go.yaml` the `sata_ssd` should probably use `platform: drivetemp`, not `platform: acpitz`
* [ ] In `fan2go.yaml` the sensor values are like 40 or 80 (°C?), should be in milli-degrees (e.g. 40000)
* [ ] The "regex" examples for "platform" are confusing. Some of them are just strings (which requires allowing the regex to only match part of the string - is probably the case, but should be mentioned?) and `"corsaircpro-*"` does not mean what one might assume it does - it means *"`corsaircpro` followed by 0 or more `-`"* - **not** *"`corsairpcpro-` followed by anything"* (that would be `coarsairpcpro-.*`)
  - Based on that example I originally assumed that it doesn't use real regexes but shell/fnmatch() wildcard patterns - but looking at the code, indeed real regexes are used
* [ ] HwMon fan examples (in the README and fan2go.yaml) contain
  ```yaml
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 1
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 1
  ```
  But the `fan2go detect` output has an `Index` and a `Channel` row. Is `Channel` the RPM or PWM channel? Is `Index` the other one or something different or not represented in the config at all?
* [ ] fan2go.yaml has this section:
  ```yaml
  # Configuration options for each fan controller responsible for an individual fan.
  fanController:
    # The rate to update fan speed targets at
    adjustmentTickRate: 200ms
    # Time to wait for a fan to respond to a control change
    pwmSetDelay: 5ms
  ```
  This is global, though, how does one configure this for *"an individual fan"*?
* [ ] The `controlAlgorithm:` section of the `cpu` fan has this:
  ```yaml
      direct:
        # together with maxPwmChangePerCycle, fan speeds will approach target value
        # with the given max speed.
        maxPwmChangePerCycle: 10
  ```
  What does "cycle" mean here?
* [ ] "Curves" section of the README has this comment in the sample: `# Sensor value -> Speed (in pwm)`  
  As far as I understand, that's incorrect - the speed is in the range of 0..255, but that is somehow normalized and mapped to MinPWM..MaxPWM
  - [ ] Seriously, this should go. Either use raw PWM values or percent values ("1% .. 100%" or "0.0 .. 1.0") - maybe even support both raw PWM and percent, but the current thing is just too confusing and a PITA to use (where does the user even learn what value between 0 and 255 maps to what fan speed? I think `fan2go fan init` shows raw PWM values, and only if used with `-v`)
* [ ] Explain "PID" - what does the "setPoint" represent, what are p, i, d?

DanielGibson, 2025-05-31:
They're almost all about documentation?

> HwMonController has both Name and Platform, fan2go detect prints the Name, but in the config the "platform" is supposed to be used

is more about documentation in the code (add comments to make clear what Platform and Name in the HwMonController mean), but that's still documentation - and probably only a few lines of comments, nothing big.

Ok, the point about using percent or raw PWM is not exactly documentation, but directly related to its parent-point which is about misleading documentation.
I could create an issue about that - does it need a completely new issue, or is there already one that's close enough to the topic?

markusressel, 2025-05-31:
I haven't really worked with sub issues yet but we can give it a try 👍

DanielGibson, 2025-06-01:
I fixed the most simple issues (where I knew what to write without reading much source) in a commit in #373.
I ticked those off above.