markusressel/fan2go #387: "Curves" section of the README has this comment in the sample: `# Sensor value -> Speed (in pwm)`

markusressel, 2025-06-04:


markusressel, 2025-06-04:
First of all: We don't need to be serious here, this is a pet project fan controller software, not the firmware of a pacemaker 😄 

Yes. the documentation on that is/was misleading, and is a result of the history of fan2go.
The text has since been changed to `Speed (0-255)`, to better reflect that this is not in fact directly referencing the PWM value that is written at the end of the fan controller loop (although it can be).

Changing this to a percentage value is a great suggestion and might help to understand how a curve is applied inside of the fan controller.

> where does the user even learn what value between 0 and 255 maps to what fan speed?

I think this should be understood by reading the documentation about Curves and Fans, but it seems it is not documented well enough.

By default `0` refers to the fan being at minimum speed and `255` refers to the fan being at full speed. What "minimum speed" and "full speed" mean is determined by fan2go during the fan initialization sequence and can be manually specified using the `minPwm` and `maxPwm` config options.

DanielGibson, 2025-06-04:
It turns out that this is even more complicated than "it's interpolated between MinPWM and MaxPWM": In reality it's only interpolated between MinPWM and MaxPWM if `shouldNeverStop` is set for the fan.
Otherwise it's between 0 and MaxPWM.

BTW I have a solution for allowing specifying fan values as percent (while keeping backwards compat), will open a draft PR soon :)

DanielGibson, 2025-06-04:
> Otherwise it's between 0 and MaxPWM.

I wonder if it would be better in the non-shouldNeverStop case to treat 0 specially and always map it to 0, and mapping 1-255 to MinPWM..MaxPWM.
Would make the curve behave much more similar to the `shouldNeverStop` case

Draft PR for allowing percent: #390

DanielGibson, 2025-06-06:
For neverstop it would make more sense to use StartPwm anyway, at least at first if the fan isn't spinning at all

DanielGibson, 2025-06-06:
> But since fan2go doesn't even measure the MinPwm correctly yet, MiniPwm is equivalent to StartPwm.

... unless configured manually