markusressel/fan2go #391: Nvidia changes break fan2go when libraries are missing

edt-xx, 2025-06-05:
**Describe the bug**
fan2go rebuilt on 05 Jun will not start getting error:
fan2go[227947]: /usr/bin/fan2go: symbol lookup error: /usr/bin/fan2go: undefined symbol: nvmlSystemGetCudaDriverVersion_v2

**To Reproduce**
Steps to reproduce the behavior:

rebuild and restart and have no nvidia/cuda

**Expected behavior**
it should start

**Screenshots**

**Desktop (please complete the following information):**

- Distro: ArchLinux
- grover 6.15.0-1-stable-git #1 SMP PREEMPT_DYNAMIC Mon, 26 May 2025 12:49:11 +0000 x86_64 GNU/Linux
- sensors version 3.6.2 with libsensors version 3.6.2
- `fan2go version`: Fails with above error

**Additional context**
Looks like the nvidia code is not handling error correctly (no nvidia hw here)

azafrob, 2025-06-05:
> [@edt-xx](https://github.com/edt-xx) please try again with AUR package version `fan2go-git 0.9.2.r68.g5bc5dd3-1`. Works for me now.

I'm on fan2go-git-0.9.2.r68.g5bc5dd3-1 and still get the same error

fan2go: symbol lookup error: fan2go: undefined symbol: nvmlSystemGetCudaDriverVersion_v2

markusressel, 2025-06-05:
How did you install it? Did it rebuild the binary?

edt-xx, 2025-06-06:
@marusressel rebuilt 
ls -l fan2go-git-0.9.2.r68.g5bc5dd3-1-x86_64.pkg.tar
-rw-r--r-- 1 ed users 15380480 Jun  5 22:08 fan2go-git-0.9.2.r68.g5bc5dd3-1-x86_64.pkg.tar
and it now works.  Looks like @azafrob was the one still getting an error.

azafrob, 2025-06-06:
@edt-xx I didnt clean the cached version my bad (I'm a linux noob). Now it works thank you so much to you both!

markusressel, 2025-06-06:
I have released a new version [0.10.0](https://github.com/markusressel/fan2go/releases/tag/0.10.0) which should result in an automatic clean build.

closed 2025-06-06 by markusressel