markusressel/fan2go #395: Tests Fail With Nvidia-Related Errors

mtoohey31, 2025-06-08:
**Describe the bug**

When I run the tests on a system without `libnvidia-ml.so` available, many of the tests fail.

**To Reproduce**

1. Ensure `libnvidia-ml.so` isn't available on the system.
2. Run `go test ./...`
3. `internal/{controller,curves,fans,hwmon,persistence,sensors}` will all fail with errors like the following:

   ```
   /tmp/nix-shell.t9RKCy/go-build2134731420/b001/persistence.test: symbol lookup error: /tmp/nix-shell.t9RKCy/go-build2134731420/b001/persistence.test: undefined symbol: nvmlGpuInstanceGetComputeInstanceProfileInfoV
   ```

**Expected behavior**

The tests should pass without error. It would also be fine for there to be some way to explicitly indicate that Nvidia-related functionality shouldn't be used for tests, if we don't want to risk skipping these tests accidentally on systems that should support them.

**Additional context**

I'm building from the `0.10.0` tag. I'm running into this because I'm trying to update the fan2go Nix package. With the `buildGoModule` infrastructure there, it's typical to test things after the build succeeds. We could disable all of the tests, but if it's possible to fix it so that most of them can still run, that would be ideal.

DanielGibson, 2025-06-08:
I can't reproduce this.
both `make test` and `go test ./...` run fine on my laptop (that has no nvidia GPU and no libnvidia-ml.so)

Does your build infrastructure set any options for the go compiler?
Maybe something that makes `cgo` code **link** C dependencies dynamically instead of **loading** them dynamically (with `dlopen()`)?

mtoohey31, 2025-06-08:
Thanks for the reply! I am building on a NixOS system so there could definitely be something different about my environment that's causing this. I've tried two different systems, one with an Nvidia GPU and one without.

> Does your build infrastructure set any options for the go compiler?
> Maybe something that makes `cgo` code **link** C dependencies dynamically instead of **loading** them dynamically (with `dlopen()`)? Because 1. the test code doesn't call _anything_ nvidia-related and 2. fan2go itself doesn't use `nvmlGpuInstanceGetComputeInstanceProfileInfoV()`

I don't think this is the cause because I get the following output with these commands:

```console
$ go test -c ./internal/sensors
$ ldd sensors.test
        linux-vdso.so.1 (0x00007ffff7fc3000)
        libpthread.so.0 => /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib/libpthread.so.0 (0x00007ffff7fb8000)
        libdl.so.2 => /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib/libdl.so.2 (0x00007ffff7fb3000)
        libresolv.so.2 => /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib/libresolv.so.2 (0x00007ffff7fa1000)
        libc.so.6 => /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib/libc.so.6 (0x00007ffff7c00000)
        /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib/ld-linux-x86-64.so.2 => /nix/store/g3s0z9r7m1lsfxdk8bj88nw8k8q3dmmg-glibc-2.40-66/lib64/ld-linux-x86-64.so.2 (0x00007ffff7fc5000)
$ ./sensors.test
./sensors.test: symbol lookup error: ./sensors.test: undefined symbol: nvmlGpuInstanceGetComputeInstanceProfileInfoV
```

...so the test binary doesn't appear to be linking to anything Nvidia-related. My usual build environment only enables vendored dependencies and `-trimpath`, but these errors occur either way.

If I comment out/remove a bunch of the Nvidia code, the tests do pass for me... It's very messy but I've attached a diff that seems to work. Are you sure the tests never call anything Nvidia-related?

<details>
<summary>The diff</summary>

```diff
diff --git a/cmd/detect.go b/cmd/detect.go
index ac7a964..1b2a209 100644
--- a/cmd/detect.go
+++ b/cmd/detect.go
@@ -11,7 +11,7 @@ import (
 	"github.com/markusressel/fan2go/internal/configuration"
 	"github.com/markusressel/fan2go/internal/fans"
 	"github.com/markusressel/fan2go/internal/hwmon"
-	"github.com/markusressel/fan2go/internal/nvidia"
+	// "github.com/markusressel/fan2go/internal/nvidia"
 	"github.com/markusressel/fan2go/internal/ui"
 	"github.com/mgutz/ansi"
 	"github.com/spf13/cobra"
@@ -154,72 +154,72 @@ var detectCmd = &cobra.Command{
 			printTables([]table.Table{fanTable, sensorTable})
 		}
 
-		nvControllers := nvidia.GetDevices()
-
-		if len(nvControllers) > 0 {
-			ui.Println("=========== nvidia: ===========\n")
-		}
-
-		for _, ctrl := range nvControllers {
-
-			if len(ctrl.Fans) <= 0 && len(ctrl.Sensors) <= 0 {
-				continue
-			}
-
-			ui.Printfln("> Device: %s", ctrl.Identifier)
-			ui.Printfln("    Name: %s", ctrl.Name)
-
-			var fanRows [][]string
-			for _, fan := range ctrl.Fans {
-				pwmText := "N/A"
-				pwm, err := fan.GetPwm()
-				if err == nil {
-					pwmText = strconv.Itoa(pwm)
-				}
-
-				rpmText := "N/A"
-				rpm, err := fan.GetRpm()
[... 681 lines cut ...]
-	sensor.MovingAvg = avg
-}
diff --git a/main.go b/main.go
index 9fe5495..2b06bf7 100644
--- a/main.go
+++ b/main.go
@@ -19,13 +19,13 @@ package main
 
 import (
 	"github.com/markusressel/fan2go/cmd"
-	"github.com/markusressel/fan2go/internal/nvidia_base"
+	// "github.com/markusressel/fan2go/internal/nvidia_base"
 )
 
 func main() {
 	// the following is needed to make sure the nvml-lib is shutdown correctly
 	// it will do nothing if it that lib hasn't been initialized in the first place
 	// (or initialization failed, e.g. because no nvidia driver is installed)
-	defer nvidia_base.CleanupAtExit()
+	// defer nvidia_base.CleanupAtExit()
 	cmd.Execute()
 }
```

</details>

DanielGibson, 2025-06-08:
Does this only affect the test or also the `fan2go` executable?

DanielGibson, 2025-06-08:
Hmm whatever Go does here is very weird: Apparently those functions are not loaded with `dlsym()`, but are just called by their wrappers? At least `objdump -T` lists them in the DYNAMIC SYMBOL TABLE.
But, the binary still is **not** linked against libnvidia-ml.so, according to `ldd`.
So I guess the lib is opened with `dlopen()` and if that succeeded, the functions become available and can be called, and if they're called without the lib being loaded first, that results in some kind of runtime error.

I can reproduce your problem (on my non-nvidia-system) if I set the `LD_BIND_NOW` environment variable (see `man ld.so`), like `$ LD_BIND_NOW="1" ./sensors.test`

> **LD_BIND_NOW** (since glibc 2.1.1)
> If  set  to  a  nonempty string, causes the dynamic linker to resolve all symbols at program startup instead of deferring function call resolution to the point when they are first referenced.

Do you have this or something similar set?

DanielGibson, 2025-06-08:
>  Does your example error with the same nvmlGpuInstanceGetComputeInstanceProfileInfoV symbol for the tests as my runs, and nvmlSystemGetCudaDriverVersion_v2 for the binary? I notice that when checking objdump -T those respective symbols are the first entries presented, so the errors being printed could definitely be consistent with this behaviour.

Yep, the errors I get are about the same symbols.

> it is possible the NixOS dynamic linker has strange defaults

Sounds plausible - this code works on other distros like Ubuntu and Arch Linux, so the problem most probably is in NixOS.

If all else fails, there's an open PR that (hopefully) allows disabling nvml support: https://github.com/markusressel/fan2go/pull/396

mtoohey31, 2025-06-08:
> Yep, the errors I get are about the same symbols.

Great, that's probably it then, thanks for figuring this out! I'll try to see if I can determine what NixOS is doing here. And if I can't then #396 may also be helpful for me 🙂

closed 2025-06-08 by mtoohey31

DanielGibson, 2025-06-09:
Can you check if the fix in #401 works for you?

j4ncp, 2025-06-09:
This looks like pretty much the exact same problem, so I would be surprised if #401 does not fix it...

DanielGibson, 2025-06-10:
BTW, if you're wondering what exactly NixOS does differently to trigger this bug, see https://github.com/NVIDIA/go-nvml/issues/33#issuecomment-2957361104 for an explanation