markusressel/fan2go #404: security: reduce fan2go service to least privilege

johnwbyrd, 2025-06-16:
While fan2go is a useful tool, it is at present an insecure one.  It typically runs as root, accesses hardware directly, and listens on network ports.  Further, it includes a ton of go dependencies, so many so that fan2go cannot practically be audited for system security.

As a countermeasure, I have implemented a more secure systemd service file for use with fan2go.  It "works on my machine."  It attempts to limit fan2go to only those permissions it actually needs in order to do its job, thus hopefully limiting its damage if fan2go is ever r00ted.

As a side effect of that countermeasure, I have also created a Debian/Ubuntu compatible rolling distribution of fan2go, which you are welcome to link to at https://github.com/johnwbyrd/fan2go-package/ .  Current releases are available at https://github.com/johnwbyrd/fan2go-package/tags , and can be installed with dpkg -i name-of-deb-file.deb .

I suggest starting from my fan2go.service file to create your own, or otherwise just linking to my repository, which should cover the needs of most systemd users.

johnwbyrd, 2025-06-16:
Current draft of service file is here:

https://github.com/johnwbyrd/fan2go-package/blob/debian/unstable/debian/fan2go.service

markusressel, 2025-06-17:
Hi @johnwbyrd thx again.

Thx for creating a package for Debian based distros, I will link it in the README! 💓 

As for the service file, I will take a look and see what we can adopt.
While hardening the service itself is great, I would love to get rid of the "running as root" requirement if thats is possible, but I haven't found a simple way to achieve this yet.

Btw, the default configuration should not listen to any ports, as the api, monitoring and profiling should all be turned off. If that is not the case, please open an issue about that.

johnwbyrd, 2025-06-18:
Technically, it is a Debian and Ubuntu package; I think most people who want it, will be using Ubuntu; suggest updating your readme; thanks

johnwbyrd, 2025-06-19:
Running as non-root is a good goal and I am behind it.  But it may be a difficult one, as all fan controllers by definition need hardware access.  Meanwhile, you may notice that my service implementation goes to some lengths to keep fan2go in jail, so that even if it goes haywire, there's only so much trouble it can make.  It can't setuid, it can't write arbitrarily to the file system, it can't increase its own priority or security level, etc.