markusressel/fan2go #428: Latest GIT build fails to start - 0.11.1.r5.gf99d615-1

outsidefactor, 2025-09-27:
**Describe the bug**
fan2go throws an error and fails to start:

fan2go: symbol lookup error: fan2go: undefined symbol: nvmlSystemGetCudaDriverVersion_v2

**To Reproduce**
Steps to reproduce the behavior:

1. Try to run fan2go, even to just get the version
2. Fails to start, reports "fan2go: symbol lookup error: fan2go: undefined symbol: nvmlSystemGetCudaDriverVersion_v2"

**Expected behavior**
For the service not to crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- Distro: Garuda
- `uname -a`: Linux nimloth 6.16.8-zen3-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Mon, 22 Sep 2025 22:08:18 +0000 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.2 with libsensors version 3.6.2
- `fan2go version`: crashes with error, package version in title of post

**Additional context**
Add any other context about the problem here.

markusressel, 2025-09-27:
Hi @outsidefactor ,
Could this be resolved by merging #423 ?

markusressel, 2025-09-30:
@outsidefactor 

> and there is a big gap between the latest GIT code (0.11.1) and what's on the AUR (0.9.2.r68.g5bc5dd3-1)

The AUR package is a `-git` package, which means it always builds the latest source, see:

https://wiki.archlinux.org/title/Arch_User_Repository#What_is_the_difference_between_the_Arch_User_Repository_and_the_extra_repository?

So the version number in the package is irrelevant. Always check with `fan2go version` instead. Of course that doesn't work on your end currently 😅 

>  Is there a way to patch an AUR package?

No need to patch the package, you can simply clone this repo, switch to the branch of the PR mentioned earlier, and run `make deploy` to copy a fresh build to `/usr/bin`. This should also be the location of the AUR package, so it should overwrite the installed file. Alternatively you can just run `make run` to simply run a locally compiled version, or even `make build` and run the binary yourself. I think the only thing that is required locally is the `go` package.